wled/WLED #2390: Feature Request - WiFi Reconnect

jterrell345, 2021-12-05:
**Describe the problem you are experiencing**
I tend to leave my light controllers (WEMOS D1 Mini) on all year long but I've seen a problem where they seem to lose internet connectivity and are no longer active on the network and I have to reboot them to get them back online.

**Describe the proposed solution**
Please add an optional feature that would force the controller to reconnect to the WiFi in the event that it loses connectivity to the network. The retry should b e attempted every 5 minutes.

wernerhp, 2021-12-30:
Also getting that the ESP8266 disconnects from WiFi and only a reboot allows it to connect again.  It had over 70% signal strength before it dropped off. I have a time controlled preset which still runs so I know the device is up and working, but the web interface is not reachable nor is it via Home Assistant.  Tried with and without WiFi sleep on v0.12.0. Rebooting router does not reconnect the device to WiFi.

blazoncek, 2021-12-31:
Use latest beta and see if it helps.
WLED reconnects to wifi automatically if it looses it.

poelzi, 2022-01-26:
I also see a lot of disconnects that are only solved by powercycle. Compared to ESPHome or tasmota, WLED is very unstable on some devices. One device for example, ESP32eth (ethernet not used), disconnects nearly every day, while others don't.
I think somewhere around 0.11-0.12 it got worse. Maybe the firmware hangs somewhere else ?

blazoncek, 2022-01-26:
Try building with most recent ESP core.

wernerhp, 2022-01-26:
Since updating to 0.13.0-b6  it's been stable for 3 weeks.

blazoncek, 2022-01-26:
I've had WiFi outage the other day (lasted 2+ hours) due to power failure and all WLEDs reconnected automatically.
I think previous versions used older core which may have improved recently.
I am building all my builds with most recent ESP core (ESP32 3.4.0 and ESP8266 3.2.0 as used in PIO) and never had issues with WiFi.

poelzi, 2022-02-12:
I can not say this. I have many esps that constantly disconnect and actually even seem to crash. The stability gets worse for me from release to release :( I think 0.11 was the most stable of all. 0.12 got slightly worse, and 0.13 crashes 2-3 times a day, depending on the usage. Sometimes it just looses WIFI but continues with the animation, sometimes it just hangs until a powercycle. It seems to crash more if it runs a playlist, but the disconnection problem hapens always.

softhack007, 2022-12-26:
No additional feedback since one year, also wifi reconnects automatically in newer versions.

closed 2022-12-26 by softhack007