wled/WLED #2391: SyntaxError

Imperattor, 2021-12-06:
### What happened?

I build the firmware from the clean archive from the github (v.12)
only uncomment the line 21: default_envs = d1_mini
Thats all.

### To Reproduce Bug

And receive Red square on basic screen app
SyntaxError: Unexpected token < in JSON at position 0

And other error on Effects tab when i click on any effects:
Error 10: Could not mount filesystem!

### Expected Behavior

no error

### Install Method

Binary from WLED.me

### What version of WLED?

2104020

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
> Executing task in folder WLED-0.12.0: C:\Users\WINPRO\.platformio\penv\Scripts\platformio.exe run <

Processing d1_mini (board: d1_mini; platform: espressif8266@2.6.2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.7 (2.7.4)    
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting wled00.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 6.8.5
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncUDP> 0.0.0-alpha+sha.697c75a025
|-- <IRremoteESP8266> 2.7.3
|-- <ESP Async WebServer> 2.0.2+sha.d1a267c
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
[... 273 lines cut ...]
Retrieved `.pio\build\d1_mini\FrameworkArduino\uart.cpp.o' from cache
Retrieved `.pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_info.c.o' from cache
Retrieved `.pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_integrity.c.o' from cache
Retrieved `.pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_local.c.o' from cache
Retrieved `.pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_malloc.cpp.o' from cache
Retrieved `.pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_poison.c.o' from cache
Retrieved `.pio\build\d1_mini\libFrameworkArduino.a' from cache
Retrieved `.pio\build\d1_mini\firmware.elf' from cache
Retrieving maximum program size .pio\build\d1_mini\firmware.elf
Retrieved `.pio\build\d1_mini\firmware.bin' from cache
Checking size .pio\build\d1_mini\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  48.3% (used 39568 bytes from 81920 bytes)
Flash: [=======   ]  67.0% (used 699600 bytes from 1044464 bytes)
======================================================================== [SUCCESS] Took 8.93 seconds ========================================================================

Environment    Status    Duration
-------------  --------  ------------
d1_mini        SUCCESS   00:00:08.931
======================================================================== 1 succeeded in 00:00:08.931 ========================================================================
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-06:
You seem to be abusing Github issue reports.
As noted earlier, please use [WLED forum](https://wled.discourse.group) or Discord for help and other support.
If you are unable or unwilling to do that, please check YouTube videos on compiling WLED from source.
Closing.

closed 2021-12-06 by blazoncek

Aircoookie, 2021-12-06:
Exactly. Your issue has nothing to do with compilation by the way, that is successful. The error you are getting is because you are trying to use ESP8266 (4M flash) firmware for ESP01 (1M flash). Use the `esp01_1m_full` environment for that

Imperattor, 2021-12-06:
> You seem to be abusing Github issue reports. As noted earlier, please use [WLED forum](https://wled.discourse.group) or Discord for help and other support. If you are unable or unwilling to do that, please check YouTube videos on compiling WLED from source. Closing.

I apologize. I may have misunderstood something.

Imperattor, 2021-12-06:
> Exactly. Your issue has nothing to do with compilation by the way, that is successful. The error you are getting is because you are trying to use ESP8266 (4M flash) firmware for ESP01 (1M flash). Use the `esp01_1m_full` environment for that

and can i activate OTA in this build ?

blazoncek, 2021-12-06:
#2040

Imperattor, 2021-12-06:
> #2040

Oh! i see (((