wled/WLED #2394: Pixel count for RGB

9ghost9, 2021-12-07:
Copied from Led preferences 
Total LEDs: 1

RGB selection set 3 gpio and all work on Channel 1 only . I believe we need 3 leds minimum or 1 pixel .
I tested 3 outputs with mosfets  and all are tied to #1. Esp32 and most recent release .

blazoncek, 2021-12-07:
You can assign PWM White and have each of the channels controlled individually (by any color or white slider). This set up will also let you have each of the channels act as segment.

This is as far as I understood your cryptic description. 😉

9ghost9, 2021-12-07:
I hope not cryptic , just a bit perplexed as to why there is 3 circles on the colors page  labeled 1 ,2 and 3 , but only 1 is assigned to 3 gpio .

As far as I can tell by testing , it is impossible to get 3 discrete channels . 
I will conclude that WLED does not provide discrete channel level control to each gpio .

blazoncek, 2021-12-07:
Ohhh! Those are not channels but color placeholders (as paint buckets).
Some effects will use just color 1, most will use color 1 for foreground and color 2 for background, a few will use all three colors.

9ghost9, 2021-12-07:
All this time I have wasted trying to understand why these 3 channels are not working . Oh my gosh .

9ghost9, 2021-12-07:
So how does one drive the gpio independently ?

blazoncek, 2021-12-07:
Assign each as a separate PWM White bus.

9ghost9, 2021-12-07:
Ok , now that I know how rgb is handled . I chose rgb x 3 which gives 9 gpio  and position the channels accordingly to the color order and all works correctly with the color wheel  . 

Thank you .

9ghost9, 2021-12-07:
A final note on this , using WLED as designed for rgb selection of 3 gpio works perfectly with DMX/E1.31 as discrete control . 

I should have tried this first not last .

Thanks again .

closed 2021-12-07 by 9ghost9