wled/WLED #2396: ESP 8266 not responding to anything except MQTT

corvus2606, 2021-12-07:
Version: 0.13.0-b5
Board: D1 Mini
Bin: WLED_0.13.0-b5_ESP8266.bin

I have had what initially appeared to be the same wifi connection issue as others with my ESP8266, however I have been experimenting today and identified that the board will still respond to MQTT.  If I send "RB" to the wled/boardname/api topic, it will reboot and for about 3 seconds the web interfaces become responsive, and then they go offline again.

I can send ON/OFF and T as well as preset selections and they all work through MQTT.

I have been trying to apply the setting to enable UDP send as this was indicated to correct the perceived wifi issues, but despite getting a success response to the JSON API request in the few seconds it's available, it never saves the setting.  Wifi sleep is disabled.

at the times when I can get a web response, the board is showing 18kb of free memory.

corvus2606, 2021-12-07:
Update: I managed to get on to the web interface for a few seconds, and was hit with this error:
[image]

From research, it looks like out of memory, but again, was showing 18kb free

dystechnic, 2021-12-07:
I have the same issue on a Wemos D1 mini. I get a slightly different message sometimes. That the connection to the lights is lost.
I found out that the webserver becomes unresponsive as soon as I enable MQTT. Disabling MQTT resolves the issue. It does not matter what other settings I change. Number of LED's, AP on or off, UDP on or off,the result is the same.
I tried different firmware versions, 0.12.0, 0.13.0-b4 and 0.13.0-b5 both from a binairy file and compiled from source.

When the webserver is non-responsive the ESP still reacts to MQTT (as corvus2606 mentioned) and ping. So the WiFi stack is not affected. 
I also do not see any memory issues in the webinterface.

dystechnic, 2021-12-09:
I did some more testing. Even with MQTT disabled and WiFi sleep disabled the Wemos is frequently unreacheable. Even from  the android app and home assistant. So it is not only related to MQTT unfortunately. With MQTT disabled the Wemos becomes reachable again through the webinterface, app and hass after some time.The interval seems random.

At this time I am running 0.13.0-b6 with 150 led's. I removed all segment config and the LEDs are powered from an external psu. Free Heap: 16.5kb, Filesystem: 49/1024kB (5%)

corvus2606, 2021-12-09:
@dystechnic are you logging you WiFi signal strength by any chance?

Mine is in an area that is not very good signal, and I seem to only have issues when it's raining which interferes with 2.4 and 5ghz signals.  When it was unresponsive, I also tested UDP sync, and it also works.  After moving the D1 a few inches to one side, I haven't had a single drop out and the web interface is staying available for the last 24 hours in spite of rain.  seems the route to my wifi AP was blocked by something.

I suspect that in times of low signal, the app prioritises MQTT and UDP for network traffic rather than the more intensive web interfaces and JSON API.

dystechnic, 2021-12-09:
The Wemos is outside, but the AP it connects to is 50cm away behind a brick wall. So it is pretty close.
The signal strength is fluctuates between 76% (-62dBm) and 78% (-61dBm) as stated in the info box in the GUI of WLED.
I will try moving the Wemos because there is a power outlet in the line of sight between the wemos and the AP.
But, I have a second one that is just laying on my desk that has the same issues and disconnects when MQTT is enabled.

dystechnic, 2021-12-11:
I've done some more experimtents and the weirdest thing happened.

The Wemos lives in a VLAN where more domotica equipment resides. I use DHCP to give it an IP address with the MAC-address coupled to an IP address in the DHCP server (192.168.30.49) . This way I know which device has wich address. I noticed today that the Wemos was stable while experimtenting on it. At that time the IP address was dynamicly assigned (192.168.30.168). 
But as soon as I locked the MAC address to the IP address (192.168.30.49) the instability was back. Removing the address from the DHCP server and setting same address in WLED the problem was still there. 
I changed the IP address in the WLED settings to 192.168.30.41 and  rebooted the Wemos. To my surprise everything was stable. The webinterface stayed reachable together with MQTT and UDP active.
I tried this with different adresses ( .42, .168, .50) and everything worked.
Recoupling the MAC adress in the DHCP server with the 192.168.30.41 produced the same stable result.

The I reassigned the 192.168.30.49 address to the wemos and is was immediately unreachable again. 
I am a network engineer by profession and traced and sniffed as much as I could during the tests but I found nothing on the network that could explain this behaviour.
I reproduced the issue with a different Wemos and was able to reproduce the result. As soon as I use the IP addres 192.168.30.49 it becomes unreachable.

I have no clue what triggers this behaviour but it seems unlikely to me that the WLED code reacts to a single IP address. There is no other device with the IP address on the network and there is no firewall involved in the setup.
It is stable for now and I will keep testing some more this weekend.

dystechnic, 2021-12-12:
Unfortunately, I'm back to square one.
After a day both Wemos devices are again only reachable  through MQTT or UDP. No webpage.
I noticed also that they are rebooting quitte a lot. WiFi reception is 90+% and stable according to my WiFi manager.
I even flashed them back to 0.12.0 but that didn't help also.

blazoncek, 2021-12-12:
You can try  updating ESP core if you are compiling from source.

corvus2606, 2021-12-12:
I gave up on the hardware I was running on and swapped it out for a spare I had which was purchased at the same time from the same supplier.  I loaded a backup of the settings and presets and updated my DHCP table to assign the same static IP, I haven't had a single drop since I replaced it.

The only real difference in how they were set up was the replacement was flashed directly with WLED_0.13.0-b5 using the web installer, while the previous one was originally flashed with WLED 0.10 using ESP8266 Flasher and OTA updated through the other versions.

I intend to experiment with the old one and try flashing it again using the web installer to see if that prevents the issues i've been having.

blazoncek, 2021-12-12:
I have noticed that not all ESPs are made equal.

corvus2606, 2021-12-12:
> I have noticed that not all ESPs are made equal.

I agree on that, but these ones are from the same batch with the same manufacturer, so there may be more at play.

blazoncek, 2021-12-12:
... even in the same (purchased, not necessarily manufactured) batch.

dystechnic, 2021-12-12:
I agree that not all ESPs are equal. But I also have a few from the same manufacturer. Bought in one batch an one of those had been runnning for 3 months without issues. After reflashing it from 0.12.0 to 0.13.something it shows the same symptoms as the other ones. Ik can not get i t back to the original situation.
All compiles where done with the latest core libs. 

It is not so much the webinterface not working that is bothering me, I can build my own wth Nodered and the API. It's the reboots that's the annoying part. 
I think I'm going to try it with an ESP32 next.

paularmstrong, 2021-12-21:
I've been having wifi issues since upgrading to 0.13.x as well. They're so bad that I can't even stay connected long enough to downgrade back to 0.12.x.

blazoncek, 2021-12-21:
I have about 30 ESPs (with WLED) in various places around my house. Various versions too.
The problematic ones are always those that have poor reception or did not have proper boot (power issue) -  not WLED version.

If I'm being cynical -  you can always use serial to downgrade. 😉 

Still, and this not being cynical, I have made some stability improvements in my fork and usually @srg74 provides various builds in his repo. They are to be tested and reported back to be then included in master.

richard-scott, 2022-02-10:
I've been having this issue with the 0.13.0-b6 release. I found that enabling the device's AP to be 'always on' has helped me even though I don't use the AP mode as it's connected to my home wifi.

blazoncek, 2022-02-10:
> I've been having this issue with the 0.13.0-b6 release. I found that enabling the device's AP to be 'always on' has helped me even though I don't use the AP mode as it's connected to my home wifi.

have you tried @srg74 experimental bins as well?

richard-scott, 2022-02-10:
@blazoncek no, where can I get them?

blazoncek, 2022-02-10:
In his repository: https://github.com/srg74/WLED-wemos-shield/tree/master/resources/experimental

richard-scott, 2022-02-10:
Is the [esp8266_no_mods.bin](https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/esp8266_no_mods.bin) the one I want?

richard-scott, 2022-02-10:
> If you have other hardware, you can also try others.

I tried that binary, but controlling my leds from Home Assistant stopped working so I have had to revert.

Wlada, 2022-03-15:
I tried with this experimental repository and still getting controller reboot very often.

D1 mini-board. ESP8266

richard-scott, 2022-03-15:
I have changed:

- AP Opens: disconnected
- Disable WiFi sleep: off

and since then I've been ok on the original firmware. Just upgraded to v0.13.1 and seems to still be ok.

Wlada, 2022-03-15:
Thanks, I give it a try.

How many LEDs do you have?
Do you have some kind of resistors on GND or capacitators?

closed 2022-03-15 by blazoncek