wled/WLED #2397: Sync not working with colors only

DarthWeber, 2021-12-07:
### What happened?

If i use sync with all options (brightness,color,effect) as default it works well. But if i disable "effect" for example, the colors do NOT synchronize anymore 

### To Reproduce Bug

WLED1 Group 1 send only
WLED1 Group 1 receive only with option effect disabled

### Expected Behavior

I expected the recevibg WLED will show a fixed effect but with the color palette synchronized from the main WLED 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13b5

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-26:
This is intentional.
Palette only synchronizes with effect but the basic colors do synchronize without effect.

closed 2021-12-26 by blazoncek

dudewheresmacar, 2023-12-05:
> This is intentional. Palette only synchronizes with effect but the basic colors do synchronize without effect.

Could this...not be intentional? It'd be really nice to be able to switch between color palettes within the current active effects across devices without having to adjust each light individually.

blazoncek, 2023-12-05:
There is no room in the UDP for such thing. And the code is already a mess supporting all of the versions back to 0.1
If we'd drop support for older versions and restructure UDP packet then maybe, but just maybe.

The best workaround is to use MQTT and broadcast palette only.

dudewheresmacar, 2023-12-05:
> The best workaround is to use MQTT and broadcast palette only.
Thanks for the reply. I'm not familiar with MQTT (or how any of this works, really hahah) are there any resources you recommend I check out?

> the code is already a mess supporting all of the versions back to 0.1
If maintaining support for years-outdated versions is limiting forward innovation, might be time...