wled/WLED #2399: GPIO22 and GPIO21 (Q1 and Q2) on QuinLED-DIG-Quad (v1) with ESP32 (TTGO T7 v1.3) does not work on WLED 0.13.0-b5 (does work on v0.12.0!)

roelbroersma, 2021-12-08:
### What happened?

Using an QuinLED-DIG-Quad (v1) with ESP32 (TTGO T7 v1.3), it doesn't work to enable GPIO22 or GPIO21 (Q1 and Q2) onWLED 0.13.0-b5 while it does work on WLED 0.12.0   (just rolled back to test it, and also needed to do a power-off/power-on after rollback).

### To Reproduce Bug

Use WLED 0.13.0-b5 and install on a TTGO T7 v1.3 on a QuinLED-DIG-Quad (v1).

I also compiled using MULTI-RELAY Usermod so I can test it with relays and PUSH buttons.  Same result but much easier to test.

To do so, the only things I have changed in **platformio.ini** are:

Added build-flags:
  -D USERMOD_MULTI_RELAY
  -D USERMOD_DALLASTEMPERATURE

and uncommented (in lib_deps = )
    OneWire@~2.3.5
    milesburton/DallasTemperature@^3.9.0

(Yes, I also enabled the temperature)

Rollback to standard release (binary) WLED 0.12.0 and you'll see that (as soon as you set the RELAYPIN to IO22 or IO21) it works.

### Expected Behavior

You would normally expect that **all** 4 outputs (IO22, IO21, IO19 and IO23) which are: Q1, Q2, Q3 and Q4 do work.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-02-12:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
See #658

closed 2022-02-12 by blazoncek

blazoncek, 2022-02-12:
Use newer version 0.13-b6 if possible.
GPIO21 & GPIO22 do work but may be reserved for I2C if you compiled such option.