wled/WLED #2401: Json api request power toggle with playlist together not working if leds are turn on

dvino, 2021-12-08:
### What happened?

Json api request power toggle with call playlist together not working if leds are turn on

[cfg .json.txt](https://github.com/Aircoookie/WLED/files/7678173/cfg.json.txt)
[presets.json.txt](https://github.com/Aircoookie/WLED/files/7678175/presets.json.txt)

### To Reproduce Bug

I need call {"on":"t","ps":203} by analog button. Where "ps":203 is the playlist. If LEDs are turn on, power toggle not working. This call just enable playlist. If "ps":203 light preset (not playlist) it works correct.

### Expected Behavior

Power toggle

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-08:
use "pl" instead.

dvino, 2021-12-09:
{"on": "t", "pl": 203} - not work mcu is freezes
{"on": "t", "ps": 203, "pl":0} - not power toggle

{"on":"t","playlist":{"ps":[1,2,3,4,5 ... "end":0}}} - works!!! 
Thant you!

closed 2021-12-09 by dvino