wled/WLED #2402: switching segments with json doesn't work right since version 0.12.0

Jojopirat, 2021-12-08:
### What happened?

I have created a desklight the first four leds should show me status of some objects, the rest should be able to switch on and off.
I realized this by the following json object:
{"seg":[{"id":0,"i":[[0,0,0,0],[0,255,0,0],[0,0,0,0],[0,0,0,0]],"on":true,"bri":100},{"id":1,"on":true,"bri":100}]}
in version 0.11.0 this works. By switching the value of "on" in the second segment the light turn on and off.
since version 0.12.0 by switching the value on the second segment, also the first segment turn on and off.
But it happens only if the json object is send a second time. In my project this happens by resend it after checking the state of status leds periodically.

### To Reproduce Bug

sending json object with two segments in with "on" value of the secound segment ist false and the "on" value of the first ist set to true.

### Expected Behavior

Only the second segment should be switched off.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.13.0-b6_ESP8266

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

with version 0.11.0 it works fine, after this version it fails.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-02-12:
Please check latest master and report back.

softhack007, 2023-01-06:
@blazoncek can we close this issue?

blazoncek, 2023-01-07:
I think yes. Either I do not correctly understand the problem or I cannot reproduce.

closed 2023-01-07 by blazoncek