wled/WLED #2410: Setting of led count ignored on stripe growth

zbx-sadman, 2021-12-11:
### What happened?

I was connect led stripe (50 led) to the controller, create preset and set it ID as "the boot preset". All worked ok, and after physical power off/on cycle worked too.
Then i added more (+50) leds to the stripe, changed "led count" setting (100 led), tap "save cofig". All worked until physical power off/on cycle was performed. After that whole "new stripe part" stopped working. And all other manually choosen effects worked on "old stripe part" (50 first led) only, "Led count" setting showed new stripe led count (100 led)
After that i was changed number of leds in the boot preset ('stop:' to 100 led), save it, and repower controller. Now whole stripe (100 led) works after power off/on. 

Seems that firmware take number of led from the boot preset, "led count" setting ignored or overwritten on the runtime config.

### To Reproduce Bug

Connect 50 led stripe to the controller.
Set "led count" setting to 50, save setting, 
Create preset and save it.
Apply this preset on boot.
Make physical power cycle (off/on),
See 50 leds worked.
Choose other effects manually and make sure it works on all 50 leds.

Add more 50 led to the first stripe.
Change "led count" setting to 100, save setting,
Make physical power cycle (off/on),
See 50 leds worked, it's ok - preset have 50 led only stripe length.
Make sure thah "led count" setting is 100
Choose other effects manually and see that it works on first 50 leds.

### Expected Behavior

I guess that manually choosen effect must processeed for "led count" pieces on the stripe, not only for "boot preset led count".

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b3 (build 2109220, SoundReactive fork) , WLED 0.13.0-b5 (build 2111170)

### Which microcontroller/board are you seeing the problem on?

Other

### Anything else?

My Controllers is: 
1) SP511e (ESP8265, 1Mb flash);
2) H803WF (ESP8266, 4Mb flash).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-11:
This is not a bug but intended feature.
If you want to update LED count in your saved preset, load it, change segment stop LED and save preset again.

closed 2021-12-11 by blazoncek

zbx-sadman, 2021-12-11:
System does not make autochange 'stop' value in the preset - i understand it.

I wrote about _ignoring global_ "led count" settings for any effect when local "led count" settings from boot preset have another value.