wled/WLED #2411: Going out of sync after a while

fribse, 2021-12-11:
### What happened?

I have three WLED controlled lights.
All running 0.12, using udp port for sync
1) esp8266 send and receive sync (star)
2) esp32    send and receive sync (facade)
3) esp8266 receive sync (candles)

I'm running a standard colorloop effect.
If I manipulate 1) or 2) they sync up, but after a while they are a tiny bit out of sync, and this worsens over time.
They are all timesynced to a local timesource (Home Assistant, chrony addon).

### To Reproduce Bug

Set up colorloop and sync between several instances, and over time they drift.

### Expected Behavior

They shouldn't drift, but keep synced

### Install Method

Binary from WLED.me

### What version of WLED?

0.12

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-11:
Use 0.13-b4 or newer and use virtual buses (network DDP). Even regular sync is handled better in 0.13
The problem of time drift (and hence effects) is due to lack of RTC on ESP chips.

closed 2021-12-11 by blazoncek

fribse, 2021-12-12:
Hi @blazoncek 
You have to enlighten me on this one, I have no idea what you are talking about 🤣 
I've updated to 0.13-b4, and I can find the word DDP in the docs, but it is only mentioned, not explained.
This morning the esp32 was out of whack with the esp8266's.
My current settings after the update is this:
[image]

blazoncek, 2021-12-12:
Use LED settings and select network type.
Please use [WLED docs](https://wled.discourse.group) or Discord for help and support questions.