wled/WLED #2421: Kill running timer upon receiving Alexa off command

LankyGit1, 2021-12-15:
If you have an API command defined that has a timer as part of its functionality eg sunrise or sunset then if wled receives an off command from Alexa there should have the ability to abort the timer. As currently the leds will switch off then continue with the running timer command for example increasing intensity.
Whether this functionality should have a switchable option of working as it currently does ie the timer will keep running or abort I don't mind.

Unless I am missing another way of doing this?

Thanks

blazoncek, 2021-12-16:
Can you be more specific what kind of timer are you talking about?

LankyGit1, 2021-12-16:
I have the following API command preset triggered at a specific time each day. If I decide to abort the preset by using Alexa to turn off, the leds turn off but then as the timer is still running on the wled interface the leds start to illuminate again. Sorry can't remember where I got this API command from originally but I think I have tweaked the colours also.

&FX=0&CL=hff0000&C2=hffb845&A=5&NL=20&NT=255&NF=2

blazoncek, 2021-12-16:
NL is nightlight which is independent of anything else. It is deprecated and may be removed in the future.
Replace it by playlists if possible.

closed 2022-02-12 by blazoncek