wled/WLED #2423: Individual Power Limits per GPIO Pin

digitydogs, 2021-12-16:
**Is your feature request related to a problem? Please describe.**
It frustrates me that i have to artificially limit parts of my displays if using multiple pins and differing numbers of leds per pin

Example use case:  Holiday lighting using a higher DC voltage center power supply to distribute power to lighting displays in yard with buck converters and WLED built into and driving the lighting display.    

WLED pin D4 driving a display with 81 leds, pin TX driving a display with 50 leds.  Each display has 3A of power available to it.  The display on the TX pin is shared with the Wemos board limiting display available current to ~2600ma.   Currently It is not possible to set it so pin D4 can use the full 3A and pin TX can only use 2.6A.  This results in having to effectively limit both strands to 2600ma (for a combined pull of 5200ma) which also means i have to lower the overall maximum brightness of my displays even further to ensure color and intensities match throughout  

**Describe the solution you'd like**
I would like to see power limiters implemented per GPIO.

**Describe alternatives you've considered**
An optional current limiter applied per segment/segment group could also be effective and would allow further flexibility in display designs.

**Additional context**
Sorry if this was already brought up, i couldn't find anything with the search. Oh yeah... Keep up the great work!

Thank you for your ideas for making WLED better!

blazoncek, 2021-12-16:
Power limiters per GPIO port will produce uneven brightness between your LED strips.
I know it may be tedious but you can achieve manual limiters by adjusting segment brightness (making each of your strips one segment).

digitydogs, 2021-12-18:
> Power limiters per GPIO port will produce uneven brightness between your LED strips.

Thats not really a concern when the strips are doing different things, are in different locations, or are being controlled by a program like xlights.  

Barring that the real issue with doing it that way is that WLED currently sees available power as being shared evenly among connected lights. 

Quick example. 2 strings of lights, one controller, 2 power sources. 

25 lights on string 1, ~1.5A at max, 6A available. 

150 lights on string 2, ~9A at max, 6A available 

Currently if i was to set max current at 12A, which is technically available, string 2 will attempt to pull full current and malfunction in one of several ways, string 1 will light up properly.  

If i reduce the current available to 7.5A and turn the brightness to max string 1 will light at 70% of its actually maximum and string 2 will malfunction as it is still pulling too much power.  (6.4A)

By the time you adjust to the sweet spot on the current limit you will have lost a large range of available brightness on both your strips as WLED has no way to know where that power is available just that it exists. 

Using segments and brightness limits to fix this is really just a band aid, and is going to require finding the sweet spot everytime something happens that erases segments, being able to set proper current limits eliminates the hassles. 

As for uniform brightness, we probably also need to add an option along with the multiple current limits to enable or disable uniform color/brightness between used pins/segments for when you need them to be the same.

blazoncek, 2021-12-18:
Duplicate of #2364

closed 2021-12-18 by blazoncek

blazoncek, 2023-11-16:
8f7f9ec3676af6d1da8a9b5114a43d53dbcc0fd7