wled/WLED #2424: Longitude entry pro lem

n1iic, 2021-12-16:


n1iic, 2021-12-16:
Greetings everyone. When trying to enter my longitude, I was unable to select a negative value for the West. The android entry screen  of only number did not have a minus symbol available, and the only option was for an Eastern location available on the time page.
I wish I could contribute to the coding for this, since it's such a great project, but I will only be able to offer these little details for the problem that I found.

Thanks 

Jason

n1iic, 2021-12-16:
One idea is to compute the +/- setting from the user selected time zone

blazoncek, 2021-12-16:
This is a shortcoming of the Android device (web view implementation).
Try updating your Android to a newer version or use computer to enter LAT/LON.

blazoncek, 2021-12-16:
> One idea is to compute the +/- setting from the user selected time zone

How to do that for southern hemisphere? They have the same problem with LAT.

Nevertheless I will think of something, though it is low on priority list.

Aircoookie, 2021-12-16:
Could instead of negative entry do radio buttons for N/S and W/E.

As a workaround, use Gboard or another 3rd party keyboard. The person at Samsung who made that keyboard unable to enter negative numbers was about as smart as the one who made the specification for log4j.

blazoncek, 2021-12-16:
> Could instead of negative entry do radio buttons for N/S and W/E.

Was my thought as well.

n1iic, 2021-12-17:
I copied the dash from a text field using the full keyboard.

Samsung is the problem. Software on mainstream devices can be a pain, that's for sure. It worked fine on my Blackvue phone.

blazoncek, 2021-12-23:
check commit https://github.com/Aircoookie/WLED/commit/2c1418105154457c658d1f01fa9c329561567cee
[image]

closed 2021-12-23 by blazoncek

n1iic, 2021-12-25:
That is great! Nice work, and thank you again.