wled/WLED #2426: Wled sk6812 power supply freezes Led

Dado19899, 2021-12-16:
Hello everyone who reads answer the problems! I'm new to the site, I have a power problem for esp32, wled installed on it. Led strip sk6812 5v 300led-pixel. It only works for me when connected with a 5v usb (1m led)cable to esp32 (I can change colors work fine) and when I connect an additional power supply  DC5v 5a (when the tape is connected it has 5.70V) The LED light strip freezes and does not react, I change the color and there is no progress when I turn it off, it remains to work on 5v usb on the esp 32 board, the power supply works normally again. I tried to put another version of wled install on esp32 but it doesn't help. I tried everything, grounding, capacitor and nothing helps  Also in the app wled I set the maximum power supply for Led and nothing reacts, is there anyone who has solved the problem.Thanks

Dado19899, 2021-12-16:
[image]

blazoncek, 2021-12-16:
Please use [WLED forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2021-12-16 by blazoncek