wled/WLED #2428: problem on preset

mascolo74, 2021-12-18:
### What happened?

I load a playlist of 20 wled on restart starting id 30 playlist on configuration perfect everything ok every 10 seconds the lighting effect changes.
I want to check a preset from the app I load preset 10 (example), preset 10 starts but the playlist immediately restarts, I can't stop the playlist and load just one preset.
wled 13.0b6  esp8266

### To Reproduce Bug

load playlist from configuration,      Apply preset 30 at boot (0 uses defaults) ok
playlist go
I check preset 5  (example) from the app
lights switch to preset 5
10 seconds the playlist id 30 restarts

### Expected Behavior

I check preset 5  (example) from the app
the lights switch to the chosen preset
I edit the chosen preset, and the lights stay on the chosen preset
save preset settings
reload playlist
restart playlist configuration

### Install Method

Binary from WLED.me

### What version of WLED?

13.0b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-18:
Please provide presets.json.

mascolo74, 2021-12-18:
explain to me how to do it

blazoncek, 2021-12-18:
http://your-wled-ip/presets.json

porterhouse80, 2021-12-21:
I'm having this same issue, but with timer controlled presets. Time of day starts a playlist, next time of day changes to a preset, however the playlist starts over 30 seconds after this preset is set, when it should just remain on the preset.

Funny thing is I have a third preset that shuts the lights off and it works, the playlist does not carry on.

Here's my presets.json
[presets.json.txt](https://github.com/Aircoookie/WLED/files/7756496/presets.json.txt)

blazoncek, 2021-12-21:
There is a bug! Thank you for reporting.
Fixed and will be available in next build.

closed 2021-12-21 by blazoncek