wled/WLED #2430: Analog PWM broken on 0.13.0-b6

lucaf86, 2021-12-18:
### What happened?

I'm using WLED on a Wemos d1 mini to control some withe leds string (classic led with two pin) using the analog "pwm white" setting in the UI. 
Everything works correctly on  0.12.0 but when upgrading to 0.13.0-b6 the leds are off. The strage thing is that selecting some effects like Aurora or Bpm the leds turn on but with a very low brightness. 
Settings are the same on both versions and I alwayes erased the entire flash before changing version. Also tried to install WLED with the online installer but on 0.13.0-b6 it doesn't work

### To Reproduce Bug

Install WLED 0.13.0-b6 and set LEDs outputs as "PWM  white" and select GPIO pin.
(I tried with bot GPIO 4, wich is the one connected to my leds, and GPIO 2 connected to integrated blue led )

### Expected Behavior

leds should turn on and the effects should work

### Install Method

Binary from WLED.me

### What version of WLED?

 0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-18:
Are you using MOSFETs?

lucaf86, 2021-12-18:
No, I'm using a led driver based on CN5711, it has VCC, GND, led output and an enable pin where I connected the pwm output.
But as I said the issue is visible also with led disconnected and gpio set to 2 to use the builtin blue led on the wemos.

blazoncek, 2021-12-18:
What is the PWM frequency you are using?

CN5711 can support 2kHz at most.

lucaf86, 2021-12-18:
yes I know that the max frequency is 2KHz but I've set WLED_PWM_FREQ   to  880, which should also be the WLED default value

lucaf86, 2021-12-19:
I did some tests with different revision:

v0.12.0 -> OK
v0.12.1-b1 -> OK
v0.13.0-b5 -> OK
v0.13.0-b6 -> not working
head (commit c27117e99e28ca8b363f4b4b81ac77e6726252fb) -> not working

blazoncek, 2021-12-19:
Find the commit that is still working and I'll have a look what could caused it.

lucaf86, 2021-12-20:
The last working commit on master is 9547ac353d018f6a5db6ffc7c27eb31df7f56e7b but the next commit is a merge request #2285 (f6e5b67f0dbf717c26aea2641496e08a2a797b05).
I tried to narrow down the problematic commit in this merge request and I've found out that the problem is in this commit:

Revision: ea0f37f5b9c52a38f72c06c3db055c1832a5bc60
Author: cschwinne <dev.aircoookie@gmail.com>
Date: 24/11/2021 11:02:25
Message:
CCT bus manager logic simplification

CCT from RGB if none set (-1)
----
Modified: wled00/FX.h
Modified: wled00/FX_fcn.cpp
Modified: wled00/bus_manager.h
Modified: wled00/cfg.cpp
Modified: wled00/colors.cpp
Modified: wled00/data/settings_leds.htm
Modified: wled00/fcn_declare.h
Modified: wled00/html_settings.h
Modified: wled00/html_ui.h
Modified: wled00/json.cpp
Modified: wled00/set.cpp
Modified: wled00/wled.h
Modified: wled00/xml.cpp

blazoncek, 2021-12-20:
Can you post details (screenshot) of your LED settings page?
I am interested in LED configuration and Auto white and CCT configuration (bottom of the page).

lucaf86, 2021-12-20:
[image]
[image]
[image]

lucaf86, 2021-12-20:
And here the settings with the new white management section on the nonbworking revision

[image]

lucaf86, 2021-12-20:
And here the settings with the new white management section on the non working revision

[image]

blazoncek, 2021-12-20:
Try CCT additive blending 100%.

lucaf86, 2021-12-20:
nothing changed. Tried with 100% and other lower values but it doesn' work.

blazoncek, 2021-12-20:
I have tested PWM output using IRFZ44N and SE2300 MOSFETs on ESP8266 and ESP32 and it is working as expected (with CCT and without, ie PWM White & PWM CCT).
Unfortunately I do not own CN5711 to test with it but I guess if you are using it with ESP8266 the frequency of 880Hz should be ok so I think it may be some other configuration (or incorrect use) that is causing problems.

Please also try to use different kinds of Auto white calculation or verify White slider (not color wheel) in UI.

Closing.

closed 2021-12-20 by blazoncek

lucaf86, 2021-12-21:
@blazoncek I found the problem in the configuration, with v0.13.0-b6 and above the PWM white led settings works only if the 
white color is selected in the color wheel and Color 1 is set in the palette. with older revisions instead it worked with any color selected.

Followinng my working settings on v0.13.0-b6 in case anyone has the same problem:

[image]
[image]

blazoncek, 2021-12-21:
You can use appropriate Auto White mode to achieve similar behaviour as in previous versions.

Joshfindit, 2022-03-06:
I’m also having issue with analog PWM.

I’m on b7 on the D1 mini, and have a fringe setup:
1 segment of 7 Warm-White-only APA102s using GPIO 0 and 1 as suggested in the documentation
1 segment of a single 5mm Warm White analog LED using GPIO 4

I’m not able to test different comments as I don’t have a setup that’s able to compile, but I’ve measured voltage on the PWM pin and it’s outputting 0.07V whether it’s on or not.

Joshfindit, 2022-03-06:
@blazoncek Could you explain what you mean by “appropriate Auto White mode”?

blazoncek, 2022-03-06:
> @blazoncek Could you explain what you mean by “appropriate Auto White mode”?

Look into LED settings page. Advanced.