wled/WLED #2431: Requesting time zone

trimmkm, 2021-12-18:
**Is your feature request related to a problem? Please describe.**
My time zone does not appear in the drop down list of time zones.

**Describe the solution you'd like**
I would like the "Atlantic Standard Time" (GMT -4) to be added to the list of time zones. 

**Describe alternatives you've considered**
I have successfully used the UTC offset to correctly set the time on my devices, but would rather see my time zone listed.

**Additional context**
Excellent firmware, thank you!

Thank you for your ideas for making WLED better!

FreshImmuc, 2021-12-20:
Just use GMT time zone with offset -4

trimmkm, 2021-12-21:
I am currently using GMT with an offset of -14400 seconds, which gives me the correct time.  Is there a place that I should enter this in hours instead of seconds?

Regards
Kevin

signout, 2021-12-21:
> I am currently using GMT with an offset of -14400 seconds, which gives me the correct time. Is there a place that I should enter this in hours instead of seconds?

No, you're doing it right. The offset is in seconds.

blazoncek, 2021-12-22:
You can make appropriate entry in ntp.cpp, test it and submit PR when it behaves correctly.

blazoncek, 2022-02-12:
No response from OP with details of TZ info or PR submitted.
Please use WLED [forum](https://wled.discourse.group) or Discord for futher help and support questions.
See #658

closed 2022-02-12 by blazoncek