wled/WLED #2432: Preset names, ID, total count in the JSON API

groto27, 2021-12-19:
I am using Node Red with WLED and want to be able to start a random preset from my list of presets. Is it possible to add the count of existing presets to the JSON API along with the name and ID? Currently I believe we just see the ID the currently set preset. I want to generate a random number with Node Red based on the number of presets I have configured (so I need the count that I have configured). If I am missing something please let me know. Thanks!!

blazoncek, 2021-12-19:
Not possible unless you download /presets.json and parse it yourself.

deggle, 2022-01-18:
Just randomly seen this - but this can probably be closed as the response from @blazoncek is actually the answer I think. From Node-RED you will be able to request `http://[wled]/edit?edit=%2Fpresets.json` and easily access the name and ID as requested.

Here's a flow that grabs it for me, and it would be very easy to iterate and select one:

`[{"id":"e80ccbdf8aacdf94","type":"inject","z":"b934e239.09224","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":410,"y":4600,"wires":[["f534acecefc13eeb"]]},{"id":"4bfb94548c8e85db","type":"debug","z":"b934e239.09224","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":750,"y":4600,"wires":[]},{"id":"f534acecefc13eeb","type":"http request","z":"b934e239.09224","name":"","method":"GET","ret":"obj","paytoqs":"ignore","url":"http://10.0.1.235/edit?edit=%2Fpresets.json","tls":"","persist":false,"proxy":"","authType":"","senderr":false,"credentials":{},"x":570,"y":4600,"wires":[["4bfb94548c8e85db"]]}]`

Another option would be to just store a set of WLED state JSON in an array inside a template node and send a random one to the WLED API (e.g. instead of on the WLED device).

blazoncek, 2022-01-19:
No need for /edit prefix.

blazoncek, 2022-02-12:
Due to the nature of how presets work in WLED (i.e. they are never stored in RAM) this may not be implemented any time soon.
Closing.
If you feel that WLED code has changed dramatically, since this post, and allows preset storage in RAM, you can reopen the issue.

closed 2022-02-12 by blazoncek