wled/WLED #2434: sorry there was an issue loading your presets

HuguesGauthier, 2021-12-19:
### What happened?

lost all my presets

on the preset tab i get this message
"sorry there was an issue loading your presets".

### To Reproduce Bug

dont know

### Expected Behavior

keep my preset

### Install Method

Binary from WLED.me

### What version of WLED?

v.0.13.0-b4 build 2110110

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-19:
If your browser does not show a backup of your presets there is not much that can be done.
You should backup your presets before doing changes.
Go to Config -> Security & Updates and you'll see backup button.

demikl, 2021-12-22:
Hi.

I'm facing the same kind of issue on my setup (v0.13.0-b6 on ESP32), and I was able to reproduce it so that someone could investigate.

First, restore this preset backup :
[presets_new_with_orange.json.gz](https://github.com/Aircoookie/WLED/files/7764218/presets_new_with_orange.json.gz)

Then, add the preset n°22 (the last one I recorded) to the playlist n°8 named "Noel". This step triggers the bug.

When looking at the JSON fetched by the UI through http://x.x.x.x/presets.json I've spotted two inconsistency:

- the content of preset id 8 (playlist "Noel" that was just saved) is duplicated ; the updated one is at the end of the JSON document, but the previous one is still found in the middle of the JSON document
- the JSON document is invalid, since the previous preset id 8 named "Noel" does not have the key` "8":{...}` anymore :

[image]

Looks like the issue is in the JSON serialiser or deserialiser. I'm OK to submit a PR to solve this bug, but I've never looked to the code of WLED. Can anyone please give me a hint on where to start with ?

blazoncek, 2021-12-22:
I have about 50 presets and never encountered such issue.
Were you perhaps saving too fast? Or reloading page in the middle of saving?
This could explain what happened.

Apart from that the attached JSON is prettified, WLED does not save prettified JSON and will not handle it properly. Use minimized JSON.
Also try to avoid national characters in names (it should not matter but nevertheless).

demikl, 2021-12-23:
I just tried the same steps, but with a minified JSON on the first step (restoring presets). That's far better, no more issue when addind a preset to the playlist.

Thanks for the hint.

goofus, 2021-12-23:
Same issue here with WLED_0.13.0-b6_ESP32.bin

TC72, 2022-10-31:
I just had the same issue and noticed a quick error message pop-up when I reload the page. In Chrome using the inspector to check the javascript console I could see the error was linked to incorrect JSON format.
I downloaded a back-up of the presets and could see straight away a load of extra characters had been added to the file.

For me the fix was to edit the file and import it again and then everything started working again.

The problem is it would keep trying to load this file every time I restarted and creating new presets based on the current settings kept creating a corrupted JSON file. So it's difficult to get out of the error loop.

blazoncek, 2022-10-31:
File system corruption (includes presets.json) is a known issue that occasionally happen.
There are no reliable steps to reproduce, sometimes it may happen often while saving presets other times it may not.
0.14 includes possible mitigation though it may still not be 100% as the root cause is still unknown.

blackketter, 2022-11-24:
Just hit this myself while trying to create set up a new hardware configuration.  I can copy some data (nice feature) but can't seem to clear the error condition or delete the bad data.  Do I need to do a factory reset?  I'm on v0.14.0-b0 "Hoshi"

blazoncek, 2022-11-24:
Use any public JSON formatter/parser to help find the error.

blackketter, 2022-11-26:
> Use any public JSON formatter/parser to help find the error.

Thanks @blazoncek, but it wasn't just the JSON being malformed, it seemed that the filesystem was corrupt or something.  Couldn't clear the error or make new presets.  A factory reset fixed it.

natekinkead, 2022-12-09:
I'm having this issue too.  Multiple times on a single ESP32 board and now also on a Dig-Quad v3.  I'm a web developer, so I inspected the presets.json file that it was trying to load, and I saw this.
[image]
I think it always happens while I'm using the Android app.  I always have to go to my computer, get the presets.json, clean out the junk that was injected and restore them with my cleaned file.  It would be really nice if someone can figure out why the JSON is getting extra data injected like this.

blackketter, 2022-12-09:
Here's some more data, I hope it may help.

It happened again to me, using Safari on macOS.  I was able to extract the corrupted file, edit out the extra noise and reupload to work.  

Interestingly, the corruptions were all long strings of 0x20 space characters, between presets (like on the screenshot above).  There was also one at the beginning, after the very first open brace character.   I'll attach the corrupted file. 

[wled_presets.json.txt](https://github.com/Aircoookie/WLED/files/10198697/wled_presets.json.txt)

It happened when I was making a bunch of changes in quick succession via the web interface.  I wonder if there were overlapping requests that stomped on each other.   Another possibility is in the code that may assemble presets together.  Just some wild guesses...

calumr, 2022-12-23:
I think the problem is bufferedFindObjectEnd. It's not handling the case where there is a space before the object begins:

`"9": {
    "on": true,
    "bri": 120,
    ...`
    
It gets called with f.position() pointing at the first \<space\> before the opening '{'. objDepth is set to 0 initially aso the `while (count < bufsize)` loop only gets run once. 

https://github.com/Aircoookie/WLED/blob/3905cad68d9eb6d2f8b7b7ebbc99ab23d5eed358/wled00/file.cpp#L132

calumr, 2022-12-23:
I think that fix works. It's tricky to reproduce, so here's my cfg & presets:

[cfg.zip](https://github.com/Aircoookie/WLED/files/10295848/cfg.zip)

Re-saving the "Green glitter" preset was enough to trigger the bug for me. With the fix, I can re-save any object and nothing gets overwritten.

blazoncek, 2022-12-23:
> I think that fix works.

Unfortunately it doesn't.

GmGuy91, 2023-01-04:
> I think that fix works. It's tricky to reproduce, so here's my cfg & presets:
> 
> [cfg.zip](https://github.com/Aircoookie/WLED/files/10295848/cfg.zip)
> 
> Re-saving the "Green glitter" preset was enough to trigger the bug for me. With the fix, I can re-save any object and nothing gets overwritten.

I uploaded your "presets.json file into one of my many esp32 boards that are having issues, and it fixed it, so now I will go thru and do the same to my others, and edit from there

natekinkead, 2023-02-12:
I installed WLED 0.14.0-b1 on a new board and I was having this issue again, but instead of a bunch of junk characters getting injected between objects (like before), this time it was removing the key from before the object that I was saving, and it was adding a new object at the bottom with the key.  Having one object missing a key causes invalid JSON.

After a great deal of troubleshooting, I finally figured out a workaround.  After I cleaned up my JSON and I was ready to import it, I first needed to minify it.  You can use an online too like this https://www.cleancss.com/json-minify/ to minify your JSON.  Then, you can upload the minified version when you restore the presets and now it won't break.

@Aircoookie , I think a good fix for this issue that should be easy to implement would be to take the incoming "restore presets" JSON and run it through a "minify JSON" function in case people upload a formatted version like I was.

callimero, 2023-03-04:
I also suspect the android app. Shuffled some segments and bang.

nexizen, 2023-03-07:
Preface: I am not a programmer. I have been dealing with this issue for ages. I literally gave up on one of my setups and decided to run a single effect because I couldn't get it to save any presets. My issue may be different from what others in this thread have been seeing, but I thought it would be worth sharing what I found. (Note this was all done on a fresh install of 0.14.0-b1.)

I uploaded calumr's presets and they worked. Everything loaded and the presets worked as expected. Then I tried to modify them. I tested a few different changes.  (All I had to do was navigate to settings and then back to see the failure message.)

1. Overwrite an existing preset.
2. Delete an existing preset.
3. Create a new preset.
4. Modify a newly created preset.

1 and 2 both failed, but 3 and 4 worked fine. I could add new ones all I wanted, but any change to the existing ones would break.

Here are the changes I noted in the JSON when modifying the existing presets.
1. Deleting an existing preset would remove the preset number for the first line of it's section, but the rest would remain.
[image]
2. Overwriting an existing preset would remove the present number from the first line of the section (like above) and then the entire body of the preset configuration would be re-written outside of the existing section.
[image]

My final test was to hand-create a JSON and test that. I created a preset on top of the ones from calumr. I saved down the preset backup. I gutted it, leaving only my preset. I then uploaded that.... Now everything works as expected. I can create new, overwrite, delete, everything. I'm so confused. 

I was able to cause some other issues with saving presets, but I haven't been able to consistently recreate them so far. I'll keep poking at it.

Phobos-7, 2023-11-22:
Unfortunately, this issue wasn't fixed in the latest 0.14.0.
I am not using an Android app, only a web interface.
I got "Sorry, there was an issue loading your presets!" error right after I backed up my presets to a json file...

lornz, 2024-02-22:
Issue still exists in 0.14.2-b1.

I attached the JSON of my presets file. Whenever I click the "save" button on one of these presets I get the message
"Sorry, there was an issue loading your presets!"

The JSON itself looks alright to me.
[wled_presets_backup_220224.json](https://github.com/Aircoookie/WLED/files/14377037/wled_presets_backup_220224.json)

blazoncek, 2024-02-22:
use [jsonformatter.org](https://jsonformatter.org) to validate

lornz, 2024-02-23:
> use [jsonformatter.org](https://jsonformatter.org) to validate

I did. The JSON is correct.

natekinkead, 2024-02-23:
The JSON has to be minified.

blazoncek, 2025-12-29:
Anyone stumbling on this: Presets on ESP32 get corrupted (99.999%) on save by incompatible flash mode used when flashing ESP for the first time.
To resolve the issue *you will need to flash new bootloader using DIO or DOUT mode*.