wled/WLED #2436: Audio Client

FreshImmuc, 2021-12-20:
Add audio reactive modes. I know there are already some github forks with audio reactive stuff but they just dont work or just work with a microphone.

Add a feature to "record" or "share the screen/window" with the web-interface chrome-tab on pc and make the music you shared appear as effects on the LED strip.

Something like tap on an effect that is audio reactive, it prompts you to share your spotify window or youtube window and then takes the audio from that shared window and sends it to the chip.

Stektax, 2021-12-20:
This would be AMAZING to have omg

FreshImmuc, 2021-12-20:
Ok so I just found LedFx. THIS IS SO AMAZING. THIS COMBINATION IS SO GREAT

Thank you for making this, this is so great

closed 2021-12-20 by FreshImmuc