wled/WLED #2438: 0.12.bin Time setup after reboot not refresh...

ddijanic, 2021-12-21:
I made i hwtronics/clock-7seg-fw-WLED  0.12.bin is version.
Its works , I am happy, very happy.

one small issue.

after the power outage, when it turns on again, the time is the year 1970 .1.1 00h: 00m: 00s and it starts counting from that time.
I waited and 1h will not pull myself out of the NTP server

if I enter the wled ui, change the color, the time is set. or run a preset, the time is set, or I turn it off and on via the web and, the time is set...

I tried to make a macro, with 00h and 01 min, run color, run preset, run off, and on .. AND THEY DIDN'T DO THEIR TRICK ...

only if I click on pc or mobile.

IF i go to time, sad disane NTP server, m and enable back on, Noting time is 1970. i only come back if I change color, effect, preset, turn off / on

do you have any ideas?

blazoncek, 2021-12-22:
Use 0.13

ddijanic, 2021-12-22:
the clock-7seg-fw-WLED is on 0.12... i am making clock...

blazoncek, 2021-12-22:
Update to 0.13.

ddijanic, 2022-01-02:
Give me .bin or link for 0.13 clock 7seg..

Aircoookie, 2022-01-02:
Make sure your device has an internet connection and can reach the `0.wled.pool.ntp.org` NTP server. You can also ask @dinamitemic to rewrite the 7seg code as a usermod we can compile in using 0.13 :)

dinamitemic, 2022-01-02:
Hi all, sorry if I do not replied before but I don't have anymore so much free time as before for my projects :(
However, I taken a look at the usermod folder and I've seen two seven segment mods that I think are working as my overlay option.
If someone could test them we can merge everything together and I can update/remove the instruction on my website in order to do not make confusion and standardize the process for using WLED with almost every 7-Seg like clock.
Always happy to see how this project is continously growing!! :D
P.S. Happy new year to all of you!!

ddijanic, 2022-01-04:
@dinamitemic 
i can test on my watch, no problem.
 just write to me. I use esp6288 4 mb

blazoncek, 2022-02-04:
For enhancements in usermods, contact usermod developr and have him/her submit updated usermod.
Closing.

closed 2022-02-04 by blazoncek