wled/WLED #2441: Fixed number of leds on ESP-01

Natal-CWB, 2021-12-23:
### What happened?

I can't change the number of LEDs. On this new version the quantity of Leds are fixed at 30.
Before I was using version 0.12 and it was possible to select the amount of leds.
[image]

### To Reproduce Bug

Install WLED_0.13.0-b6_ESP01.bin on any ESP-01 and look at LED Preferences

### Expected Behavior

enter the number of leds

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.13.0-b6_ESP01

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Natal-CWB, 2021-12-24:
I also tried it with 0.13b5 and 0.13b4 and both didn't work, only 0.13b3 worked.

blazoncek, 2021-12-24:
You should have read release notes.
The field is no longer required. You change number of LED in the bus/strip data.

closed 2021-12-24 by blazoncek