wled/WLED #2442: Transition with Colour not working from Home Assistant

Morphy99, 2021-12-24:
### What happened?

I can't seem to get WLED to transition colour changes over a specified time from the HA integration.

Tried with transition off in WLED too. Interestingly adding brightness to it does seem to correctly transition that over the specified amount of time but colour still changes at the WLED default.

Using WLED v0.12 and HA 2021.10.7

### To Reproduce Bug

Sending this in HA dev tools:
```
service: light.turn_on
target:
  device_id: 1801e557652a4728b36641238515a00b
data:
  transition: 60
  rgb_color:
    - 255
    - 255
    - 255
```
Just uses the default transition in WLED.

### Expected Behavior

Colour would transition over the specified time.

### Install Method

Binary from WLED.me

### What version of WLED?

v0.12

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-24:
Use 0.13-b6 and report back. We are unable to fix older releases.

Morphy99, 2021-12-24:
Thanks but still the same with 0.13-b6

I've updated the post with a better explanation (I hope!) and the version tested.

blazoncek, 2021-12-24:
use "tt" instead of "transition" in JSON

Morphy99, 2021-12-24:
> use "tt" instead of "transition" in JSON

Thanks. I'm not sure how to implement your suggestion though as I'm not very experienced in using json. Can I send this from HA? I've had a look in the wiki to send via curl which I can try if that's what you want me to do?

I've got as far as this but I'm not sure what I'm doing with the segment part?

`curl -X POST "http://xxx.xxx.xxx.xxx/json/state" -d '{"on":"t","v":true,"transition":"20"}' ""seg":[{"single":{"0": [255,0,0],"20":[255,0,0]}}]" -H "Content-Type: application/json"`

I'm guessing there may be an issue with my formatting as I'm getting:

```
curl: (3) bad range specification in URL position 6:
seg:[{single:{0: [255,0,0],20:[255,0,0]}}]
```

I have 20 LED's in this case and changing to Red over 20s.

blazoncek, 2021-12-24:
Use `curl -X POST "http://xxx.xxx.xxx.xxx/json/state" -d '{"on":"t","tt":"20"}'  -H "Content-Type: application/json"`

Include "v" only if you need complete state as a result.

Aircoookie, 2021-12-31:
Hmm to really gauge if this is an HA or WLED issue, we would need to know the exact JSON command(s) sent to WLED by the integration using your dev tools command above.

Before sending the HA command, was the light on or off? If off before, there should be no color transition at all, only the brightness transition.

The difference between `transition` and `tt` is just that `tt` applies only to the current request. `transition` should also always work, so there likely is an edge-case bug with color transitions somewhere.

wildman199, 2022-01-14:
I’m also having this problem with Hoobs and Homekit.

blazoncek, 2023-01-25:
Transitions have been completely reworked in 0.14.

closed 2023-01-25 by blazoncek