wled/WLED #2446: Data pin on esp2866 incorrect i tried using 4outputs which shouldnt be a problem but in the wled configuration led output.I set the gpio to 4 but on the data pins on the esp it came out from d2 so i tried using others again i set the gpio to 2 but on the esp board the output is coming out from data pin 4
and it kept happening even when i set the gpio on the wled to gpio 5 the output comes from data pin 1 so im really confused what happened.

Capboyxd, 2021-12-25:
I [image]

blazoncek, 2021-12-25:
Use the documentation from manufacturer to map GPIO to pins marked on your development board.
Most often GPIO0 is mapped to D3, GPIO2 to D4, GPIO1 to TX and GPIO3 to RX.
Closing.

closed 2021-12-25 by blazoncek

Capboyxd, 2021-12-26:
why in the app i can only do 3output when it can do 4[image]

blazoncek, 2021-12-26:
This is due to stability issues (a design choice).
You can compile yourself by adding -D WLED_MAX_BUSES=4 to have possibilty to assign 4 GPIO.