wled/WLED #2458: Segment reset in web interface

barbatrukko, 2021-12-28:
### What happened?

I have 2 stripleds, the first of 300 LEDs connected to LED1 out and the second of 200 LEDs connected to LED4 out on a dig quad.
I start with one segment of 500 led, so i change in 2 segment, one of 300 and on eof 200, but after a random time wled web interface reset segments again to one of 500 led

### To Reproduce Bug

Create 2 or more segment, and play some effect and&ore create preset, after a while interface clear segments and create one only witl all led.

### Expected Behavior

The number of segments created must remain unchanged

### Install Method

Binary from WLED.me

### What version of WLED?

WLED w.0.13.0 b6

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I m using DIG QUAD board with ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2021-12-31:
Hi, to preserve your segment config between reboots, please save a preset: https://kno.wled.ge/basics/faq/#my-segments-are-gone-after-a-reboot

If your board randomly reverts to one solid orange segment, it sounds like a reset/crash.
If it happens when you load a certain preset, the reason is that just one segment was configured when you saved that preset. You'll need to overwrite the preset with 2 segments configured. If you anticipate changing segment lengths in the future, I recommend unchecking "Save segment bounds" for all but your boot preset.

blazoncek, 2022-02-12:
Closing since no updates form OP.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
See #658

closed 2022-02-12 by blazoncek