wled/WLED #2462: Manage color ordering by LED offset, rather than GPIO.

justfalter, 2021-12-30:
**Is your feature request related to a problem? Please describe.**
Yes. 
I have two strings of LEDs with different color ordering (RGB and BRG) chained together from the same GPIO. I want them to apply color and effects to them as a single segment, but the colors are always wrong on one of the strings because the color order config is tied to the GPIO.

**Describe the solution you'd like**
I want to be able to set LED color order based on LED offset, rather than by GPIO.  This is similar to what can be done in xLights, where you can specify color ordering on a per-model (string) basis. 

- Remove the `Color Order` option from `LED outputs`
- Add a new section to `LED Settings` for specifying color ordering. 
  - Drop-down for choosing the default color order (ex: RGB)
  - Similar to segment management, allow addition of color order overrides for LED ranges (start, length, color order). 

**Describe alternatives you've considered**
* Moving one of the strings to a separate GPIO is not always possible. In my case, there is 30 feet between the esp32 and the first string. If I moved the second string to a separate GPIO, it'd mean I'd have to run a 40 foot cable, which would be pushing it for range, as well as wasteful.

* Per-segment color ordering (#1428) is not viable as I want the two strings on a GPIO with different color ordering to operate as a single segment, not separately.
* Additional reasons why per-segment color ordering is no good:
  * It feels like the wrong place. Segments are for effects/presets, not for lower-level color order mapping. Some presets might have 10 segments, while others will have just one.
  * Managing segments across multiple presets can be tedious.  

**Additional context**

Current approach I'm using (and unhappy with) requires me to define separate segments, and manually remapping the color palette from RGB to BRG for the strings of LEDs that have the different color ordering. This is very frustrating as any changes to the palette requires me to remap colors on the individual segments, which is very tedious. 

I went so far as to implement per-segment color ordering, but quickly found that it was not in the right direction. Segment configurations do not survive power cycles, unless they've been saved to a preset. And, again, managing segments across many presets is tedious.

blazoncek, 2021-12-30:
Custom compile with:
-DCOLOR_ORDER_OVERRIDE
-D COO_MIN=<start_pixel>
-D COO_MAX=<end_pixel>
-D COO_ORDER=<color_order>

justfalter, 2021-12-30:
> Custom compile with:
> -DCOLOR_ORDER_OVERRIDE
> -D COO_MIN=<start_pixel>
> -D COO_MAX=<end_pixel>
> -D COO_ORDER=<color_order>

I should have provided a bit more detail, but I actually need to override the color order for _multiple_ ranges of LEDs, not just one. I'll update my original message.

blazoncek, 2021-12-31:
For your requirements (a very unusual) you'll need to modify bus_wrapper.h and ajust COO in such a way to be compatible with your strips.
Use COLOR_ORDER_OVERRIDE aas a guide.

Aircoookie, 2021-12-31:
@embedded-creations perhaps this should become part of `Mapping`, though I'm not sure how much additional overhead it would incur.

embedded-creations, 2021-12-31:
@Aircoookie this seems like a hardware issue,  and an ideal software fix would be to allow each bus to alternate between different LED color orders (or even different types of LEDs, e.g. RGB and RGBW).  That seems like a lot of work for a very rare case and I don't think it should be a high priority.

I don't think Mapping is the right place to fix it, but it would be possible to hack in a fix with a custom Mapping class without modifying the core of WLED, but it would still require custom compiling WLED to add the class in a Usermod.  I don't think swapping colors should be a standard feature of Mapping.

justfalter, 2021-12-31:
FWIW, I'm currently working on a set of changes that attempts to address this issue in the manner that I proposed. I'm in the middle of something at the moment, but I'll push a branch up before I get too deep into the weeds on the configuration/html/js side of things so that folks can take a look.

Thus far, the modifications sit within the core of WLED, the changes sitting in/behind `BusManager`.

justfalter, 2021-12-31:
As mentioned, here's my WIP branch for the color order modifications I've got, so far: https://github.com/Aircoookie/WLED/pull/2463

PR #2463 (2022-01-25): Allow overriding of color order by LED pixel range.
Adds a means for overriding color order for ranges of LED pixels. Default color order is still managed at the GPIO level.  This is meant to fix #2462. Up to 10 color order overrides defined.

- [x] Use `ColorOrderMap` to hold color order information.
- [x] Serialization of config to/from filesystem.
- [x] Modify `handleSettingsSet` to take a configuration
- [x] Modify `xml.cpp` to serialize settings for front-end
- [x] Modify LED settings page to allow for configuration of color order map.

### Performance: 

I figured watching the FPS would  [... cut]
closed 2022-01-25 by Aircoookie