wled/WLED #2464: Parallel LED order settings

sauer-m, 2022-01-01:
Hi,

didt find a real answer in the threads.

Is it possible to set up parallel stripes with one data line to act like one linear segment like in the pictures below?
(So, for better understanding - a running light from 1 to 80 in target image would start and run simultaneously on LEDs 30/31, 29/32, 28/33, ...).

Also I thought I could address a part of a stripe (e.g. LEDs 80-100) as separate segment (so I could light them up individually).

[image]
[image]

As for using a plain ESP8266, I would prefer to not split the 3.3V Signal for parallel addressing and/or using a level-converter (due to circuit size) 😬

blazoncek, 2022-01-01:
Please use [WLED forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2022-01-01 by blazoncek

sauer-m, 2022-01-01:
> As for using a plain ESP8266, I would prefer to not split the 3.3V Signal for parallel addressing and/or using a level-converter (due to circuit size) 😬

Oh, sorry 😌 
I created a [thread](https://wled.discourse.group/t/settings-for-parallel-u-shaped-led-stripe/4573) in the forum, thanks :)