wled/WLED #2472: Latest release not allowingmy to set the total number of LEDs

AC271, 2022-01-07:
I didn't see anything about this in the issues so I apologize if I just scrolled by. I was able to set total number of LEDs the other day and as I was troubleshooting some other issues, I noticed that I am no longer able to adjust the total number of LEDs and it's defaulted to 30.

Anyone had this issue and found a fix? I tried to re-download the latest release in case there was a corruption issue, with the same results.

AC271, 2022-01-07:
I found an article saying I have to update the hardware setup length and that seems to have worked but I'm still confused why the option was available before and it is no longer without any updates.

blazoncek, 2022-01-07:
You should read release notes.
The option became obsolete and useless.

closed 2022-01-07 by blazoncek