wled/WLED #2474: Using analog read (potentiometer for brightness) kills wifi

jkatton, 2022-01-08:
If a potentiometer is used to control brightness, as soon as it's enabled, wifi stops working.

My guess is that as per https://github.com/esp8266/Arduino/issues/1634 , the WLED seems to be polling the analog input too frequently.  I'm not sure exactly why, but the function around line 223-ish of button.cpp seems to be not restricting the polling to every 250ms as it seems to want to do.  

Removing "if (b+1 == WLED_MAX_BUTTONS) " from line 224 of button.cpp fixes the issue (at the cost of messing up any other buttons I would assume).  I don't know enough to fix things properly, but hopefully this is enough information to lead someone else to  what I would think is probably a very simple solution.

blazoncek, 2022-01-08:
Looking again at code there is an error limiting analog read frequency.

blazoncek, 2022-01-09:
Possible fix in [#aba4dc7c50ac2a25e15a673ff7bd7c3197657eaa](https://github.com/Aircoookie/WLED/commit/aba4dc7c50ac2a25e15a673ff7bd7c3197657eaa)
Please report back.

jkatton, 2022-01-10:
Seems to have fixed the issue completely, thanks!

closed 2022-01-10 by jkatton