wled/WLED #2476: sACN multicast receiver does not work well

xhpohanka, 2022-01-09:
### What happened?

Using WLED in sACN muticast mode works but it is really slow, many packets are probably dropped so the control is no smooth. If switched to unicast mode the control is fine and smooth again.

### To Reproduce Bug

- install binary
- set up wifi
- switch control to multicast
- observe 

### Expected Behavior

smooth control of brightness

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.13.0-b6_ESP8266.bin

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xhpohanka, 2022-01-09:
I have tried artnet and it behaves the same. I also tried to downgrade up to 0.9.1 again with the same behaviour.
I don't think it is a local network issue, because other clients seems to work well.

NormenL, 2022-01-25:
I had the same problem on esp32 eth, 0.13.0-b6, It worked well on 0.12 version, got it up 60fps playback on both eth and WiFi. But I have problems compiling some user mods on 0.12 and there is no rgb+CCT on it

xhpohanka, 2022-01-25:
> What did you use to send sACN and Artnet?

ETC Eos, QLC+, sACNView, all the same

mxklb, 2023-06-08:
Only brightness or other DMX channels as well? I experience something similar, see #3239

xhpohanka, 2023-06-19:
> Only brightness or other DMX channels as well? I experience something similar, see #3239

It was brightness or let's say channel value. I was not using the effect mode...