wled/WLED #2479: analog potneiometer changes Segment 0 brightness instead of Global brightness

jkatton, 2022-01-10:
### What happened?

when a potentiometer is hooked up to A0 to control brightness, the value for Segment 0 brightness is modified, instead of global brightness.  

[image]

### To Reproduce Bug

 hook analog potentiometer to wled, go nuts on the slider, wheeeee!

### Expected Behavior

I would expect the analog input should control global brightness, so that it functions if segments are being used.  Currently if any segments are being used, the analog input is ignored.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.0-b6 (build 2112080)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Thanks for all your hard work!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-01-10:
https://kno.wled.ge/features/macros/#analog-button

closed 2022-01-10 by blazoncek

jkatton, 2022-01-10:
Thank you very much for your quick reply, that solves my issue completely,
in a really comprehensive way.  Sorry that I didn't see that option earlier.

Thanks again for all your hard work, I really appreciate it.

-  Jesse.