wled/WLED #2483: Save partial presets natively

fusionstream, 2022-01-12:
**Is your feature request related to a problem? Please describe.**
I connect multiple led strips individually to pins (and used to connect them in series). Each strip lights up a different section of my area. I would like to save and recall presets only for certain areas instead of recalling a full preset.

Removing the portion of code in the preset JSON that affects the non targeted sections seems to work but; it requires manual editing of code. This is non trivial and even for me requires me to run it through a prettifier first.

**Describe the solution you'd like**
When saving a preset, it should only save the segments which are selected (checked). For me, this is extremely intuitive as I come from a live events background (e.g. Barco e2); but I believe this should be somewhat intuitive for everyone else as well.

In this case, having a "(De)/Select All Segments" button would also be great as it is useful for other things as well.

**Describe alternatives you've considered**
Recalling a full preset with secondary options to not apply to certain segments but this is quite bad from a ui/ux perspective.

**Additional context**
Scenario: I have led strips for shelving, underdesk, and computer lighting; all connected to the same esp32 running wled. I would like to have my presets (potentially) only affect these logical sections of light.

IMHO, the described solution would be the best from a ui/ux/programmer/nonbreakingchange view.

blazoncek, 2022-01-12:
This has already been considered but is non-trivial.
At least in the current state of UI.

fusionstream, 2022-01-13:
Having not looked at the code yet, would a viable solution be to (on the surface), simply check if a segment is "selected/checked"?

I wouldn't mind doing a PR because even at the worst case scenario, I should be able to hook into the produced JSON and then modify it easily. Again, I haven't looked at the code

blazoncek, 2022-01-13:
That would be one of the solutions but it would break backwards compatibility or other intentional functionality.
You can use that on your fork.

fusionstream, 2022-01-13:
What considerations would I have to take into account? Is there documentation of legacy or intended functionality?

blazoncek, 2022-01-13:
Currently all segments are stored in a preset, so anything expecting full set of segments may fail.
I have already implemented your proposal and will test it later. May not get into master though.

blazoncek, 2022-01-13:
Tested saving partial presets and it works... With a twist or in an unexpected way.
The palette or colors overwrite palettes/colors on every selected segments if the saved segment was 0.
This will require handling only @Aircoookie understands ATM.

fusionstream, 2022-01-14:
Hmm. That is unusual. Thank you. Will await the results

blazoncek, 2022-01-18:
I have this working now, but requires full segment syncing patch available in a separate branch.
May be included in one of next releases.

closed 2022-01-18 by blazoncek