wled/WLED #2485: Draging downwards in AP mode causes sitereload

Phunkafizer, 2022-01-12:
### What happened?

Dragging downwards causes site reload.

### To Reproduce Bug

Connect to WLED in AP mode using mobile phone (Pixel 3 w/ chrome)
Try to drag position in colorwheel downwards (to cyan)
OR scroll down the effect list and try to scroll up again
-> site is reloaded

### Expected Behavior

Do not reload page, just scroll & drag

### Install Method

Self-Compiled

### What version of WLED?

0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ChuckMash, 2022-01-12:
I have had the same issue when opening the web interface via the captive portal webpage in mobile browser.
My workaround has been to add "the 4.3.2.1" IP address to my WLED instance list in the app, which can be used while in AP mode.

blazoncek, 2023-08-05:
This is an issue of WLED mobile app, not WLED itself.
Please open an issue in WLED app repo.

closed 2023-08-05 by blazoncek

ChuckMash, 2023-08-05:
This might not be a WLED app issue, it happens to me when using browser on mobile to load the WLED instance IP address.

Seems like it could be addressed pretty simply with [this.](https://www.matuzo.at/blog/2022/100daysof-day53/)
Just some additional CSS.
`html, body {
  overscroll-behavior: none;
}`

blazoncek, 2023-08-06:
@ChuckMash I've tested a few phones and tablets (not a lot as there are countless models) and all quality ones do not exhibit such behaviour.
As such I deem that to be a mobile device shortcoming or poor implementation.
Still, if you have tested the above proposal and it does work on all/your devices then please make a PR.