wled/WLED #2486: Submasters or light groups - A subdivision of master

fusionstream, 2022-01-13:
**Is your feature request related to a problem? Please describe.**
I have multiple leds strips lighting up different physical sections of an area. It would be cool to have a single toggle to turn off and on all segments within an area. Presently, each segment in an area must be toggled individually. This however has the potential to break as certain segments within an area may be already be off or on depending on the scene.

**Describe the solution you'd like**
Similar to the master toggle, a sub-master toggle to override the on/off state of a selection of segments instead of all of them

**Describe alternatives you've considered**
External state checker to store present state and then programatically turn off or on each segment

**Additional context**
It's mostly a neatness and redundancy thing. If the 3rd party state checker goes offline, having this feature would allow one to directly control this grouping via WLED itself.

fusionstream, 2022-01-13:
If anyone comes across this and is looking for a quick workaround, it is to keep all relevant segments ON but to set brightness to 0 where applicable.

blazoncek, 2022-01-13:
Create a preset that will toggle segments in question. Use JSON API.
Pleas use [WLED forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2022-01-13 by blazoncek

fusionstream, 2022-01-13:
This would require manual editing bypassing the UI though. At the end of the day, integrations with things like Home Assistant will still require modifications to incorporate this functionality (e.g. presently, in Home Assistant, you have to manually select each segment to toggle). It's pretty much a UX thing.

blazoncek, 2022-01-13:
All this is achieved using JSON API which is exposed to any integration.

blazoncek, 2022-01-13:
JSON does show last/current state, use "v":true in your API call.
As said earlier create preset(s) with JSON API and use that preset(s) (from HA or button or whatever) to toggle segments.

As for UX, you are welcome to create UI/backend that will allow this functionality natively and submit a PR.

fusionstream, 2022-01-14:
Yes but this method requires third party logic.

I'll have a look at this and see how I can adjust behaviour for a PR.

Thank you.