wled/WLED #2493: Priority of live packets

coolinger, 2022-01-18:
I just replaced my raspi/hyperion+windows/hyperion-forward setup in favor of esp32/wled+windows/hyperion, and added LEDfx to windows as another "live supplier". I fully understand that I have to disable hyperion before enabling LEDfx and vice versa. But I still wanted to ask if there is a way that one live supplier can override another.

LEDfx has the E1.31 priority setting, which could be used for this. 
I guess a "newest-first" approach could be painful on UDP packets coming from the same host?

Thanks for this wonderful project!

mxklb, 2023-01-24:
Could you please build & test this PR https://github.com/Aircoookie/WLED/pull/3052 and provide some feedback? ++  reference to https://github.com/Aircoookie/WLED/issues/768

The PR Code let you set WLED device E1.31 priority to explicitly restrict control to just one of multiple controllers.

coolinger, 2023-01-27:
Thanks for your work!

I can confirm that LEDFx (E1.31) now overrides hyperion (DDE)! I haven't done more testing since I gave up on LedFX for now.

mxklb, 2023-11-09:
can be closed?

blazoncek, 2023-11-09:
@coolinger please close if the issue is no longer relevant or fixed.

closed 2023-11-09 by coolinger