wled/WLED #2494: Shelly 1 loaded with WLED

To0wnn, 2022-01-18:
### What happened?

Not responding to SW (GPIO5) when conntected to L 

### To Reproduce Bug

Connect SW(GPIO5) to L

### Expected Behavior

Switching the relay / Responding to the input

### Install Method

Binary from WLED.me

### What version of WLED?

WLED beta release 0.13.0-b6 

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-01-18:
Have you set the relay pin in LED settings (not switch/button pin) to GPIO5? you may need to reboot WLED once as well

To0wnn, 2022-01-18:
Yes I changed the settings and did the reboot. Also changed the relay and is working with the GPIO0
[image]

To0wnn, 2022-01-18:
Also tried it with a different shelly 1 so it is not broken. I do run on 230VAC do not know if this is the issue for WLED?

To0wnn, 2022-01-18:
Tried with it in the 12VDC config same result also not responding to SW

Aircoookie, 2022-01-18:
Ah, now I get what you are trying to do :)

Firstly, change Button 1 type from `Pushbutton` to `Switch`
Then save and return to the main UI. Make two presets. For both, uncheck `Use current state` and enter `T=0` and `T=1` as the API command.
Now remember the ID of the preset (number displayed left of the preset name) and enter it in Time setting as the short and long action of button 1.
Afterwards, the switch should work!

To0wnn, 2022-01-18:
Tried it what you said but I think it is just not detecting the input somehow. Also i want to connect a pushbutton not a switch.

Aircoookie, 2022-01-21:
Hmm, you could also give the `Push Inverted` type a try, if the pin goes HIGH if the button is pushed.

To0wnn, 2022-01-21:
Tried that as well. But then the brightness constantly changed or something.. So that is not solving it unfortunately

softhack007, 2023-01-06:
Loses due to inactivity - stale since > 6 month.

closed 2023-01-06 by softhack007