wled/WLED #2495: Hue Integration causes intermittent resets

Majawat, 2022-01-18:
### What happened?

After so many seconds, the lights will turn off for a split second, then turn back on to the default Orange color, no matter what color/brightness/effect it was prior. After it does this, it acts normally until the next random reset.

Setup: I have a Dig-Uno running WLED 0.13.0-b6 for under cabinet kitchen lighting with 76 WS2812B 30 led/m. It is synced to a Hue light (#28), every 2500 ms. Running on Wi-Fi. Previously, I was running a ESP8266 and this was occurring on older WLED versions as well.

I've tried the following troubleshooting steps:

* Replace the power supply. Originally an Alitove 5V3A wall wart. Then a MeanWell 5V/5A wall wart. And for troubleshooting, an MeanWell 5V/60A box. The troubleshooting PSU is known working, I was previously using it for my house exterior before moving to 24v.
* Change LED strip. Unplugged the cabinet lighting and plugged in a different, known working 
* Re-flash WLED. During my previous attempt at troubleshooting, I re-flashed from scratch. (I think I still have the problem 
* Changing to a 5A fuse (not that I thought this would help)
* Removed the barrel connector and used ferrules to connect power
* Saw that the MeanWell wall-wart was providing 5.12v. The MW _box_ is outputting exactly 5v. 
* Noticed that during the reset, I'll lose a single network ping.

At this point, I've changed everything yet still have this problems.

### To Reproduce Bug

1. Configure Hue Light to sync to poll a light.

Was able to verify this by wiping the configuration by re-flashing the DigUno, letting it sit for >24hrs with no issues, then integrating with Philips Hue and immediately start having the issue.

### Expected Behavior

The WLED firmware doesn't reset and go back to default Orange color but stays synced to Hue Light

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-01-21:
Have been seeing this as well since a long time ago.
The available heap memory slowly decreases until the device resets randomly or upon trying to access the UI due to the low memory condition.
The polling must be causing a memory leak either in the TCP stack, AsyncClient library, or handling code.

Nikita240, 2022-01-23:
+1 I'm also affected by this

bobness09, 2022-03-04:
I also can reproduce this easily, when I switch on Hue Sync. With sync off it runs stable. Most of the times it crashes between 10sec and 2-5mins.
Had no problem on 0.12.0, the bug startet to occur when I updated to 0.13.0-b5 and is still on b7.

bobness09, 2022-03-15:
Still a problem in 0.13.1 for me. It crashes more often the lower the sync time is.

franciscoangelini, 2022-10-02:
i try sync my esp wled with hue and said Hue status: Invalid light ID

wusapa, 2024-01-30:
I also have the problem that my WLED keeps restarting when I sync it with my Hue lights. Are there any fixes in sight for this?

Version: v0.14.1

GroveGuard, 2024-07-13:
Hello fellas,
i struggled a LOT with these restart problems till i finaly get it is the hue connect setting. I have that problem in all of my 8 controllers (many different brands) and also in a seperate vlan. i tried nearly everything to get this working cause i have a full hue setup in my house and would love to controll all lights with the hue app, so i need that sync badly.

Is there any chance some off the genius programming minds in the wled community could figure out a fix for that memory leak problem?

if i could assist in any way pls dont hesitate to let me know.
Greetings

softhack007, 2024-07-13:
> wled community could figure out a fix for that memory leak problem?
> 
> if i could assist in any way pls dont hesitate to let me know. Greetings

Actually it might be improved already in the latest releases of 0.14.4 or 0.15.beta. there were memory leaks in both in UDP and the asyncWS library. However you'd need a build with the new esp32 framework version 2.0.9, so you'd need the ["WROVER" bin](https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b3/WLED_0.15.0-b3_ESP32_WROVER.bin) or use an esp32-S3.

The WROVER firmware needs a reduced filesystem partition, so you cannot update by OTA.

If you still see the same problem, please inform us. Personally I don't have a HUE system, so the only help I can offer is hunting for mem leaks by visual inspection of the source code. And I'm not particularly good with JSON.

GroveGuard, 2024-07-17:
Thanks for the reply. I tested all versions till 0.15.0-b3 but, when i activate the HUE sync it begins to run unstable.
I cannot connect my controller directly to the PC cause i use this [ATHOM-Controller](https://de.athom.tech/blank-1/wled-esp32-rf433-music-addressable-led-strip-controller).

Is there a way to Flash a Wrover-Binary OTA?

I just saw there is a b4-release. I will test it within the next days and let you know if it works.
Have a good one

Zugdar, 2025-04-05:
Just put together my first WLED setup with a Dig-Quad. Not a programmer but dabble in many smart systems. This still seems to be an issue 0n V0.15.0 I had resets between 4-8 minutes and have a much more stable system after disabling and removing all Hue communication. More testing/waiting to be done to verify the fix fully. Was noticing the colors shifting to the booted orange and going back to the hue bulb it was assigned to shortly after pairing the system.

Other thing I did which may or may not be a factor.
- Updated via Home Assistant to 0.15.0. Did not do any resets after that update. Might consider that if there is some sort of bug from the upgrade process.