wled/WLED #2496: MDNS doesn't work when AP is set to always on

DevDroid42, 2022-01-19:
### What happened?

If you set the AP opens behavior to Always MDNS doesn't work meaning the device can't be discovered by the wled app.

### To Reproduce Bug

Connect esp to WiFi, turn AP opens to always, observe that MDNS stops working.

### Expected Behavior

MDNS should work on the connected wifi network when the AP opens setting is set to always on.

### Install Method

Binary from WLED.me

### What version of WLED?

0.12.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-01-19:
This has been discussed elsewhere (can't recall issue # or location) but has been decided it is unresolvable with the TCP stack on ESP.
If you want to have mDNS then do not enable persistent AP mode.

DevDroid42, 2022-01-19:
good to know. It's probably worth noting this in the UI if it can't be fixed.

blazoncek, 2022-01-19:
you may want to try 0.13-b6 which has newer ESP core if there are any changes.

jmerc77, 2023-07-21:
noticing a similar issue with 0.4.0b1 hoshi. running a custom compiled version on the esp32-s3 (no code changes, just compiler configs). i say "similar" as it works fine on a desktop pc running a chromium based browser (brave in my case) or stock chrome. but never on my phone. on the phone it doesn't work in the app or any of the browsers (not even chrome/chromium based). in all cases, the dhcp assigned ip works fine as long as you know the ip. though, have not verified if changing away from always open fixes it or not because i dont know which one to pick that will still allow an ap connection regardless of weather the router is still connected or not.

jmerc77, 2023-07-21:
well, well, well. seems to not have been a WLED problem. apparently, it works now for me after noticing my phone all of the sudden refused to connect. which then i reset the router, before the reset fully took place it started working! i then had to wait for the router to reset afterwards but still worked once everything was back online. the version i was using is 2 betas old but they somehow fixed this issue on or before the version i'm using (0.14.0-b1 hoshi, latest at this time is 0.14.0-b3).

softhack007, 2023-10-02:
Seems like the issue can be closed now.

closed 2023-10-02 by softhack007