wled/WLED #2498: Autosave Usermod doesnt work correctly after toggle preset

utech-git, 2022-01-20:
### What happened?

The usermod autosave works properly when I change color, brightness, effect, etc. And it also works when I press the power button on the web interface.

But I use a Switch on GPIO5, I set a preset of "toggle" (T=2) on preset 1 and on the macros page, I set "Preset 1" as the action on switch "on -> off" and "off -> on".

So, when I press the switch, I wait 15 seconds (the default time to autosave usermod save state) but it doesn't work. The new preset (after toggle) is not saved. It seems that autosave usermod doesn't acknowledge that the power state is changed on Toggle preset

### To Reproduce Bug

Flash a binary with usermod autosave enabled. Create a toggle action (T=2) on preset 1

Press the preset 1 button to turn off the leds, wait 15 seconds, cut power then put power back on... the led return turned on, instead of turned off.

### Expected Behavior

Autosave usermod should aknowledge that the leds were turn on/off 15 seconds after pressing the toggle preset button

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-01-21:
Changing preset will not trigger auto save as well as using a preset will not.
If you are using presets this turns Autosave off.

This is intentional.

If you think this should be considered change the issue to feature request.

utech-git, 2022-01-24:
Ooohhh I see!

Since I am using WLED on home assistant, I managed to create an automation that solves this for me.

In case anyone stumble upon this in the future, here's how I did (remember to check if identention is correct)

On configuration.yaml

```
shell_command:
  wled_update_state: /usr/bin/curl "http://YOUR_WLED_IP/win&PS=250"
```

On automation.yaml
(remember to change the name of you light entity)

 ```
 - alias: WLED Autosave Changes
    trigger:
      - platform: state
        entity_id: light.wled
        to: "on"
        from: "off"
        for: "00:00:15"
      - platform: state
        entity_id: light.wled
        to: "off"
        from: "on"
        for: "00:00:15"
      - platform: state
        entity_id: light.wled
        attribute: effect
        for: "00:00:15"
      - platform: state
        entity_id: light.wled
        attribute: brightness
        for: "00:00:15"
      - platform: state
        entity_id: light.wled
        attribute: rgb_color
        for: "00:00:15"
    action:
      - service: shell_command.wled_update_state
```

On WLED LED PREFERENCES menu, set 250 on:
Apply preset 250 at boot (0 uses defaults)

closed 2022-01-24 by utech-git