wled/WLED #2500: Expose per-segment color space info to JSON API

Aircoookie, 2022-01-20:
**Is your feature request related to a problem? Please describe.**
UI and external integrations cannot determine the supported colors spaces/models of configured busses on a per-segment level.

**Describe the solution you'd like**
A JSON array, as lightweight as possible while still being concise, that makes available the following information:
- Whether a bus (partly) driven by that segment supports RGB color
- Whether a bus in the range of that segment supports a white channel
- Whether a bus in the range of that segment supports color temperature (CCT)

In case a bus supports CCT, it usually also supports at least the white channels of the 3 main colors, which control the brightness of the light.

An initial draft proposal for this functionality stores this info in an byte `lc` (light capability) per segment, e.g.

Bit | Capability
|---|---|
0 | Reserved
1 | Segment supports RGB color
2 | Segment supports white channel
3 | Segment supports color temperature
4-7 | Reserved

Therefore:

`lc` value | Capabilities
|---|---|
0 | None. Indicates a segment that does not have a bus within its range, e.g. because it is not active.
1 | Supports RGB
2 | Supports white channel only
3 | Supports RGBW
4 | Supports CCT only, no white channel (unused)
5 | Supports CCT + RGB, no white channel (unused)
6 | Supports CCT (including white channel) + RGB

Note that CCT is controllable per-segment, while RGB color and white channel have 3 color slots each per segment.

```json
{
  "info": {
    "seg": [
      {"lc": 3},
      {"lc": 1},
      {"lc": 4},
      {"lc": 0}
    ]
  }
}
```
Above the proposed JSON structure. `{"info":{"leds":{"lc":[3,1,4,0]}}}` would be more lightweight, but wouldn't make it possible to add more per-segment info in the future, and also doesn't make it as clear to an API user that `lc` is a per-segment value.

**Describe alternatives you've considered**
Always display the color wheel, even though it has no effect on the bus configuration (e.g. segment that controls only CCT busses).
This does not allow for a user interface that can dynamically show only the controls that actually apply to that segment.

**Additional context**
In case of CCT lights, adding configurable values for the coldest and warmest Kelvin values achievable by the connected LEDs is also desirable, so that color temperature can be matched to 3rd party lights with reasonable accuracy. Whether this should be possible on a per-bus or only per-instance level (to save memory), depends on the likelihood of users wishing to drive multiple CCT strips with different temperature ranges from the same instance. This is only a side note here though and not part of this request.

Aircoookie, 2022-01-20:
**Internal implementation notes**
Check which busses are part of each segment to determine its capability byte:
Should it occur every time `/json/info` is requested (double for loop, iteration count `max num segments * max num busses`)
or every time busses or the segment is updated (extra byte of memory per segment)?

New `seg` array as proposed above will require an additional 544 bytes of JSON buffer on 16 segment ESP8266.

frenck, 2022-01-21:
The problem I was trying to highlight a couple of times now, is that this isn't correct:

[image]

Because WLED allows for setting it up:

[image]

So that means, it can be configured as CCT only, the hardware is limited to CCT only, but it isn't reflected in any of the capabilities exposed.

blazoncek, 2022-01-21:
You have to consider an option that I may create configuration with RGB(W) and PWM/CCT then create presets that utilize RGB(W) only (no PWM/CCT bus in segments) and different presets that only use PWM/CCT and still third set of presets that utilize both RGB(W) and PWM/CCT.

Aircoookie, 2022-01-21:
> The problem I was trying to highlight a couple of times now, is that this isn't correct

It is. A PWM CCT busses brightness is determined by three scale factors:
- master brightness
- segment brightness
- white channel value

If the white channel wasn't taken into account, CCT busses would not support effects at all. In the current implementation, effects can modulate the brightness of CCT busses on a per-pixel level, therefore simple effects like Candle and Heartbeat work with them. Also multiple CCT busses per segment are possible, with individual brightness, but shared temperature control.
Just the color temperature is set on a per segment level, white channel is _always_ used for brightness.

I apologize if this point has caused confusion, though I think it is documented appropriately:
> CCT handling
WLED starting with version 0.13.0 also supports bus types with two white channels [....]
The overall brightness of the white channels is determined from the auto-white calculation outlined above, and as such is identical in behavior to that of single white channel busses.

frenck, 2022-01-21:
I probably don't understand it enough to be able to support it.

The issue is that the device can be configured as a CCT only device. Yet, it exposes itself as RGBW capable, while the hardware connected doesn't support colors at all.

I fail to get the problem across correctly apparently and at this point I lost the energy for it as well. I don't care, I don't use it.

Unsubbing from this issue as well, I won't get follow-up notifications.

blazoncek, 2022-01-21:
Typical.

Aircoookie, 2022-01-21:
I can understand that he has issues with the way it is currently implemented, hence this improvement proposal.
Still I am a bit disappointed he is not currently interested in finding a solution together.
Maybe he will reconsider once enough HA users wish for CCT support.

I'm not even capable of understanding what the issue really is - of course the brightness for CCT needs to come from somewhere and the white channel is by far the most flexible solution.

Still going to implement this proposal to lay the groundwork should he reconsider, or other integrations want to use CCT. Can be used to hide the color wheel in the main UI as well, in case the segment only contains CCT busses and the auto white mode is Manual.

Aircoookie, 2022-01-21:
>  Yet, it exposes itself as RGBW capable, while the hardware connected doesn't support colors at all.

No. It would expose itself as capability 6, white brightness + white temperature. No RGB, only the W component of RGBW is considered.

frenck, 2022-01-21:
@blazoncek Is there anything you want to say? If so, as that sounds like a personal response. If so, I'm not of the accepting kind and would take that rather offensive.

> Still I am a bit disappointed he is not currently interested in finding a solution together.

Nothing in this issue is an issue for me at this point. So, if you created this issue for solving the issue I was trying to report then you unfortunately misunderstood. 

Nevertheless, the proposal would be a nice improvement, as one could mix different kind of strips, now multiple strips can be connected (and thus the existing `info.leds.*` is probably (possibly) incorrect.

As I said, I probably am not capable of getting the issue across.

> No. It would expose itself as capability 6, white brightness + white temperature. No RGB, only the W component of RGBW is considered.

Setting this:

[image]

Returns this in this the API:

[image]

But the strip configured doesn't support color at all. That is the issue. It is returning the device is color capable, while the user explicitly configured it isn't.

☝️ That will cause a color wheel to be shown, but it isn't color capable.

Aircoookie, 2022-01-21:
Thank you for your response.
I agree that the behavior of `info.leds.rgbw` specifically IS in fact a bug, this value is implemented to be true if there is any bus using the white channel, but documented as being only RGBW specifically.

The reason is that prior to PWM CCT bus support, any bus supported by WLED that supported white also supported RGB, which is now no longer the case.

To align with documentation, I will change the implemented behavior of `info.leds.rgbw`. This will limit its practical usefulness and deprecation should be considered, but this proposal is a superior solution anyways, since it allows this information to be accessed on a per-segment and not only on a per-instance basis.

fishbone-git, 2022-01-28:
Does it make sense to spit out the bug specifically and track it separately?  Per the discussion on Discord, HA 2022.2b0 blocks all WLED instances with any reference to CCT.  This puts all of the WLED Athom bulbs on the block unless they are limited to RGBW only.

blazoncek, 2022-01-28:
As specified above any and all WLED devices are RGBW+CCT capable and enabled.
Regardless of hardware attached.

Aircoookie, 2022-04-16:
Implemented since 0.13.0-b7 and documentation added to https://kno.wled.ge/interfaces/json-api/#light-capabilities :)

closed 2022-04-16 by Aircoookie

jottr, 2024-03-25:
Linking the forum thread created by @justintime https://community.home-assistant.io/t/re-evaluate-cct-with-wled-and-the-ha-wled-integration for posterity.