wled/WLED #2503: Not Compatible with esp8266ex nodemcu v3

MrFr33man123, 2022-01-22:
Hi,

I musst have gotten a special board since it has not the standard esp8266 but esp8266ex.

The only working wled version ist the 0.8.2. Since i would love Home ASsistant support wich was added in 0.8.4 i would like to build the most recent version with platformIO. 

But I don't know what i have to change so that it compiles.

Right now I only get the following message:

` ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld`

This is the board.

The Specs and Pinout is here: [Datasheet ESP6288EX](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf)

[image]

If anyone has an idea how to make it compile the right way? Please let me know.

srg74, 2022-01-28:
Actually ESP8266EX is a standard chip for most of the boards. WLED should work without problem. Please come over to WLED Discord to discuss your board.

blazoncek, 2022-01-28:
It has been determined that the culprit may be PUYA flash used on those devices.
Hardware issue.

MrFr33man123, 2022-01-28:
As you can see in the Picture it is XMD Flash. Not PUYA.

[](https://datasheet.lcsc.com/lcsc/1811072025_XMC-XM25QH32BHIG_C328460.pdf)

But i will not follow this Problem anymore. Last working version with this Board 0.8.6 so i can use Home Assistant.

closed 2022-01-28 by MrFr33man123