wled/WLED #2504: Estimated Current only calculated when brightness limiter is enabled

Plawasan, 2022-01-23:
### What happened?

The estimated current value is only populated in the GUI/data exposed to HA when brightness limiter is enabled in LED settings. Since this value can be used to power consumption calculation, it would make sense (to me at least) to always calculate this value regardless of brightness limiter setting. I realize that as a workaround I can define a high enough value of the brightness limiter however that seems like an unnecessary workaround. Thanks!

### To Reproduce Bug

Disable brightness limiter in LED preferences

### Expected Behavior

Estimated current is provided regardless of brightness limiter settings

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b6 (build 2112080)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Plawasan, 2022-07-11:
Still valid

blazoncek, 2022-07-11:
Current estimation is (and will be) calculated only when the limiter is enabled as it adds substantial CPU overhead on large LED installs.
It also makes no sense to calculate the current as it is only an estimate and not actual consumption. Its primary function is to help you overcome underpowered PSUs or thin wires used for powering LEDs.

softhack007, 2023-01-12:
So it seems that it's not a bug, but intended behaviour.
Closing.

closed 2023-01-12 by softhack007