wled/WLED #2507: Locking phone with iOS app open/in background causes WLED to crash when playlist is selected.

SK360, 2022-01-24:
### What happened?

WLED has been rebooting on me sometimes when selecting a playlist. It isn't consistent and I finally figured out how to reproduce. This is running on a QinLED Diguno

### To Reproduce Bug

- Open iOS app and connect to lights
- Swipe away from WLED to put app into background
- Lock phone
- Unlock phone
- Reopen WLED which is still connected to lights
- Select Playlist

This reboots WLED for me 100% of the time. If I open the app and first back out to the light selection screen and reconnect to my lights I am able to select a playlist without issue.

### Expected Behavior

Playlist is loaded, WLED doesn't reboot.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b6 build 2112080

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

srg74, 2022-01-28:
I follow your steps and can't replicate issue. Please be more specific about iOS version.

SK360, 2022-01-30:
> 

Yeah I worked with Blaz on discourse and logged some crashes via debug mode, seems it wasn’t happy with HomeAssistant polling it every 10 seconds.  I ended up trying his fork which was solid and then I built master from this repo with the latest commits as of 3 days ago and it’s been up for 3 days not with no crashes or reboots. I’m going to close it and keep an eye on it.

closed 2022-01-30 by SK360

hi5grant, 2022-07-04:
Thanks for all the detailed info here. I’ve been seeing similar things with a QuinLED Diguno ESP32 on 0.13.0-b6 on both Home Assistant 2021.12.x (I’ve upgraded since, can’t remember the patch version) and 2022.6.8. I upgraded to WLED 0.13.1 and I get the same result. My issue is the _board_ seems to crash and is unreachable on the iOS app, lights are frozen too. 

I have another fixture with a NodeMCU ESP8266, same Home Assistant integration, 0.13.0-b6 and 0.13.1 and I can’t seem to reproduce it. Playlists trigger from HA all day long. 

I’m not expecting any kind of action from the group here. I’m just mostly reporting something weird was definitely happening that I can’t pinpoint. 

My workaround though was just to set my playlist as the default preset when booting up and I have a Z-Wave plug that I use to toggle power. Seems to work fine that way. I could see how it would be frustrating if you keep the fixture on / white most of the day but want to change it to party-mode at 6pm, my workaround wouldn’t help there; I’m just always thinking of the next weary traveler.

blazoncek, 2022-07-05:
Please use an experimental binary from @srg74 repository and test again. Please report back.
Save your configuration and presets before updating.