wled/WLED #2508: Input GPIOs should be able to be set to do nothing - MQTT report only

tazzytazzy, 2022-01-24:
**Is your feature request related to a problem?**
Not an existing issue.

**Describe the solution you'd like**
Input GPIO's such as buttons, PIR, etc, should be able to be set to do nothing, just report to MQTT only. My short term solution is to create a preset that sends the command "XYZ". This polluts the presets and creates useless items in home assistant.

I just want to report the button press or PIR motion to MQTT, and let home assistant decide what to do.

Perhaps in the "Time & Macros" menu, for the "Button actions", they can be left blank, but will still be broadcast to MQTT if "Publish on button press".

**Describe alternatives you've considered**

As stated above, I create a preset that does nothing and assign press actions to this preset. However, this creates unused presets, other than to disable any actions.

**Additional context**
Add any other context or screenshots about the feature request here.

If I know how to code your app, i'd make an attempt at a pull request. :\

blazoncek, 2022-01-24:
This defeats the purpose of buttons on WLED.
I understand what you want to achieve but the easiest solution is the one you already described.
This may not be resolved any time soon.

Aircoookie, 2022-01-25:
Another viable solution would be to assign the ID of a preset that doesn't exist to the button action.
(a high one like 240, to make it unlikely you make an actual preset using that ID later)

closed 2022-01-25 by Aircoookie