wled/WLED #2509: sp108e and sp501e

Titanius, 2022-01-24:
### What happened?

Hello,

I try to compile 0.13.0-b6 with Hardware sp501e and sp108e

------sp501e-------

with sp501e the Hardware use LEDPIN=3 fix.
Can you please set LEDPIN=3 to the only LEDPin that I can use the max pixels of 512 or as original firmware max 1024 on GPIO3. There are no more LEDPINs for DAT only for Analog RGBW

GPIO1 = Button
GPIO3 = DAT

GPIO14 = C (not Connected)
GPIO12 = W (not Connected)
GPIO13= B (not Connected)
GPIO2 = R (not Connected)
GPIO4 = G (not Connected)

-----------------------------------------------------

------sp108e-------

Then please can you add sp108e in the platform.ini

I try to use it, but the CLKPIN are not set.

and I don't know if everything is correct

[env:sp108e]
board = esp_wroom_02
platform = ${common.platform_wled_default}
board_build.ldscript = ${common.ldscript_2m512k}
build_flags = ${common.build_flags_esp8266} 
  -D LEDPIN=2 
  -D CLKPIN=13
lib_deps = ${esp8266.lib_deps}

-----------------------------------------------------

### To Reproduce Bug

use with original platform.ini WLED 0.13.0-b6

### Expected Behavior

That LEDPIN is fix, the only one, and can not change in this Hardware.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-01-24:
You can change any GPIO pin during runtime.
No need to re-compile if changing pins.

Titanius, 2022-01-24:
GPIO3 is limited.  I can't use more than ~250 LEDs here.  you can test in runtime.  LED GPIO3 and 300 LEDs

blazoncek, 2022-01-24:
Connect LEDs to any other pin. Recommended are GPIO1 & GPIO2.

Titanius, 2022-01-24:
Software Mod is easier then Hardware Mod.

`https://github.com/Operation760/SP501e-RGB-LED-Controller-/blob/master/SP501e_top_bottom_traced.jpg`

blazoncek, 2022-01-24:
You do not know what you are talking about. 😁 
We cannot reassign GPIO3 to GPIO1, but you can resolder data line from GPIO3 to GPIO1 and then change pin in LED settings.

Titanius, 2022-01-24:
I think use only GPIO3 for LED and remove the limit and remove the possibility to use Gpio1 and gpio2 as LEDPIN. I think i use the Original Firmware. Here GPIO3 works with max 1024 pixels.

blazoncek, 2022-01-24:
It is memory issue. There is not enough RAM on ESP for more LEDs on GPIO3.

Titanius, 2022-01-24:
right now we have localized the problem.  now it's up to Aircoookie if there's a chance to move the reserved memory.

blazoncek, 2022-01-24:
It is not. Easiest solution if you need more than 250 LEDs is to resolder data line to GPIO1 which is exposed as a large pad (TX).

Titanius, 2022-01-24:
then I don't understand why this controller sp501e was included in the hardware list.  You should then note in the description that only 250 pixels are possible here or is not a good choice.

blazoncek, 2022-01-24:
You are welcome to update the documentation. [WLED-docs](https://github.com/aircoookie/WLED-docs)
Fork it, make changes and submit PR.