wled/WLED #2513: Playlists no longer working

SK360, 2022-01-26:
### What happened?

Playlists will open the first preset and stop.

### To Reproduce Bug

Click playlist

### Expected Behavior

Playlist continues

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.0-b6 2201260

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Works if I flash back to WLED 0.13.0-b6 2112080

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-01-27:
you should provide playlist JSON API

SK360, 2022-01-27:
> you should provide playlist JSON API

Added above

blazoncek, 2022-01-27:
Fixed in https://github.com/Aircoookie/WLED/commit/3170fa220874524c8807f0f36a3a36f9e86cc20b

closed 2022-01-27 by blazoncek

SK360, 2022-01-27:
Rebuilt and playlists work! Thanks for the quick resolution!