wled/WLED #2514: Home Assistant: Error requesting wled data: Response payload is not completed

Resinchem, 2022-01-28:
### What happened?

This started happening around the same time as the upgrade to Home Assistant 2021.12 and the changes in WLED entities.  With over 20 WLED controllers, running versions from 0.9.1 through 0.13.0-b6, on ESP8266 and ESP32, with a couple of sound-reactive versions mixed in, I've been unable to trace this to any particular controller or version.  I temporarily disabled the 0.13.0-b6 version, considering it was added around the same time as HA 2021.12 but the errors are still flooding my log files.  None of the prior controllers or versions caused these errors prior to HA 2021.12 (I'm currently on HA 2021.12.9).

I've not noticed any WLED controllers dropping off line and all appear to function normally via either Home Assistant or the WLED web interface... it's just the flood of errors in the log (at least 50+ per day).

### To Reproduce Bug

Apparently, just upgrade to HA 2021.12.x?  This upgrade did force me to rewrite numerous automations (all in yaml), but I'm not doing anything differently or using MQTT/JSON/HTLM/API calls in any of those automations or scripts.  All these controllers (except the latest 0.13.0-b6 - which I disabled) did not cause any log errors in prior versions of HA.

### Expected Behavior

No flooding of logs with payload error.

### Install Method

Binary from WLED.me

### What version of WLED?

Multiple versions (see description above)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
(Partial... this is repeated constantly throughout the day):
2022-01-26 15:26:04 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 15:37:14 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 15:52:09 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 16:02:37 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 16:13:35 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 16:42:10 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 16:52:34 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 17:14:02 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 17:42:32 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 19:53:44 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-26 21:55:52 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-27 00:16:53 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
2022-01-27 00:33:24 ERROR (MainThread) [homeassistant.components.wled] Error requesting wled data: Response payload is not completed
```

### Anything else?

While I selected a particular install method and microcontroller above, I have a large number that all vary... again, the only commonality is that they all functioned without error prior to HA 2021.12.  If there is a way to get the logs to show an IP address or other additional information, I might be able to troubleshoot further or determine if it is a particular controller, version, etc.. I am running Home Assistant OS (I guess that's what they call the 'full' version now) in a Proxmox VM if that is helpful at all.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-01-28:
Why is this a WLED bug?
If you have issues with older versions as well consider filing an issue with HA.

Resinchem, 2022-01-28:
Honestly, I don't know at this point whether it is WLED or Home Assistant, nor whether the contributors here have anything to do with the HA integration.

But fine... I will close the issue and consider posting under HA, not that I really expect that group to be of any help based on past experience.

closed 2022-01-28 by Resinchem