wled/WLED #2519: ESP32 LittleFS was not declared in this scope

Bebick, 2022-01-30:
### What happened?

During compiling ESP32 and ESP32ETH to the current version of WLED (download 01/30/2022), the following error message occurs: 
wled00/wled.h:183:21: error: 'LittleFS' was not declared in this scope.

I am using the Visual Studio Code with version 1.63.2 and I have already done all possible updates to fix this problem.
Unfortunately without success.

### To Reproduce Bug

Compiling ESP32dev and ESP32eth with Visual Studio Code.

### Expected Behavior

A helpful explanation of how I can fix the problem.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.0-b6 (2201260)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
wled00/wled.h:183:21: note: in definition of macro 'WLED_FS'
     #define WLED_FS LittleFS
                     ^~~~~~~~
*** [.pio\build\esp32dev\src\FX_fcn.cpp.o] Error 1
In file included from wled00/cfg.cpp:1:
wled00/cfg.cpp: In function 'void serializeConfig()':
wled00/wled.h:183:21: error: 'LittleFS' was not declared in this scope
```

### Anything else?

If you need more information then just let me know.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2022-02-01:
Hi, please retry with latest master code, this should be fixed now. If it still does not work, try deleting the `.pio` directory in the main source folder.

Bebick, 2022-02-01:
Yes, perfect it works. Thank you