wled/WLED #2520: Aurora gets white background color (SK6812)

Gafielt, 2022-01-30:
### What happened?

A dimm white background is applied to the aurora effect, if the birghtness is set to high/max. This is not the case with other effects which only use a specific part of the strip at once.

### To Reproduce Bug

Use RGBW (SK6812)  and select the aurora effect with aurora color palette (or Atlantica...) and increase the brightness from low to high. At a certian point (always different) the white leds is dimmly on through out the whole strip.

### Expected Behavior

The white led should not always be dimmly on!

### Install Method

Binary from WLED.me

### What version of WLED?

v0.13.0-b6 "Toki"

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

I use an 6A 5V powersupply and set max current in software to 4A. Total Strip length is 72 leds, diveded in two times 36 leds (two outputs used). I tried if setting the autocalc white feature changes anything, but this was not the case!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-01-31:
White background is intended. The actual amount of white (RGB) depends on master brightness (and segment opacity/brightness).
If you have Auto White calculation enabled for SK6812 your W channel is calculated from RGB values (if you have Brighter mode your white may be brighter than expected).
Not a bug.

closed 2022-01-31 by blazoncek

Aircoookie, 2022-01-31:
Indeed intented. If you want to change the behavior, just replace `CRGB mixedRgb = CRGB(backlight, backlight, backlight);` with `CRGB mixedRgb = CRGB(0, 0, 0);` and re-compile.

Gafielt, 2022-02-01:
Good to know. Thanks for the quick response. Still a bit strange that the white kicks in at 100% and not at 99% with no colors selected, but i guess it's just how its calculated. I have some WS2812B wled devices and never recognized this in the effect.