wled/WLED #2521: Doesn't seem to load selected preset at boot, until I move the brightness slider.

eborchardt, 2022-01-31:
### What happened?

I created a preset that is just a dim red color. It works when I select the preset, the lights change to dim red. I go to Config | LED Preferences | Defaults and select "Apply preset [1] at boot". Then I save and reboot the ESP8622. The preset is never loaded and, furthermore, it won't change to the preset when I click the preset button, unless I first adjust the brightness arbitrarily. 

### To Reproduce Bug

See above...

### Expected Behavior

I expected the preset to be loaded on boot.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b6 (build 2112080)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-01-31:
Post a preset JSON API.

eborchardt, 2022-01-31:
Here is my Dim Red json:
`{"on":true,"bri":47,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"col":[[255,0,0],[8,255,0],[0,0,255]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

blazoncek, 2022-01-31:
What is your boot setting in LED settings?
Post screenshot?

The preset and state all look ok.

eborchardt, 2022-01-31:
[image]

eborchardt, 2022-01-31:
Actually, I ended up taking it apart this morning to see if I got the same result with an ESP32 and it worked fine with the ESP32. I just swapped the ESP8266 back in and now the issue is no longer occurring. I'm not sure how or why, but it seems there must have been a bad connection or something. It is puzzling how a bad connection could have caused it to exhibit the behavior I was experiencing, but it seems it must have been the case.

closed 2022-01-31 by eborchardt

blazoncek, 2022-01-31:
Just tested it. And I can see a brief transition from orange to red. This is expected since you have Turn LEDs at power up enabled.
It is not needed if your preset includes brightness.